igrr/esp32-cam-demo #1: Extraneous includes

krzychb, 2016-11-21:
Hi @igrr,

Basing on the includes I believe you were using https://github.com/dlbeer/quirc.
If so, did it work for you out of box or you had to tweak the code?

Waiting for the camera to arrive, I did dry run with quirc and could not make it work.
The code compiles and loads without any issues but `num_codes = quirc_count(qr)` does not find anything for s/w generated B&W QR images I provide.

igrr, 2016-11-21:
Seems like i still had `quirc` in my components directory when I was preparing the example, so didn't notice this stray reference. I suppose it can be removed for now.

Quirc has a few quirks, and unfortunately I had to modify it in a bunch of places to make it less stack hungry. I didn't have time to arrange these changes as proper patches on top of upstream version, hence QR decoding is not in the demo.

Regarding testing on generated images, quirc was pretty good in my experience. It did worse on photos, especially photos which had noticeable barrel distortion or lots of detail (=segments) on the background. So not exactly sure why it doesn't work for you. The sequence of calls looks like `quirc_begin` -> fill data -> `quirc_end` -> `quirc_count` -> (loop over QR codes) { `quirc_extract` -> `quirc_decode`.

krzychb, 2016-11-22:
This is very good news Quirc already worked for you.
I will redo my testing then once I have the camera operational with esp32.

krzychb, 2016-11-30:
Quirc indeed does not like to work out of box.

Already at `quirc_end(qr)` I am getting:

```
C:/msys32/esp-idf/components/freertos/./queue.c:1445 (xQueueGenericReceive)- assert failed!
Guru Meditation Error: Core   0 panic'ed (Unhandled debug exception)
Register dump:
PC      :  400d914a  PS      :  00060836  A0      :  800e02ef  A1      :  3ffb6730
A2      :  0000005c  A3      :  0000005c  A4      :  00000010  A5      :  ffffffff
A6      :  00000000  A7      :  ffffffeb  A8      :  00000000  A9      :  3ffb66a0
A10     :  00000000  A11     :  3f402368  A12     :  3ffb666f  A13     :  00000035
A14     :  00000000  A15     :  3ffb6675  SAR     :  00000004  EXCCAUSE:  00000001
EXCVADDR:  00000000  LBEG    :  400e09a8  LEND    :  400e09c7  LCOUNT  :  00000000
Rebooting...
ets Jun  8 2016 00:22:57
```

I will look more into it over the week end.