igrr/esp32-cam-demo #2: GPIO interrupt type error for GPIO 26 and 27

krzychb, 2016-11-21:
Hi @igrr,

I have connected the camera to ESP32_DevKitJ_V1 and flashed `esp32-cam-demo` application.
Compilation was done using the latest esp-idf release.

I have observed two issues:

1. On start up LEDC as well as GPIO  26 and 27 interrupt type errors were reported:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:0
load:0x3ffc0008,len:3592
load:0x40078000,len:5264
load:0x40080000,len:260
entry 0x40080034
I (26) boot: Espressif ESP32 2nd stage bootloader v. V0.1
I (34) boot: compile time 21:10:59
D (38) esp_image: reading image header @ 0x1000
D (45) boot: magic e9
D (48) boot: segments 04
D (51) boot: spi_mode 02
D (54) boot: spi_speed 00
D (58) boot: spi_size 01
I (61) boot: SPI Speed      : 40MHz
I (66) boot: SPI Mode       : DIO
I (71) boot: SPI Flash Size : 2MB
I (76) boot: Partition Table:
I (80) boot: ## Label            Usage          Type ST Offset   Length
D (89) bootloader_flash: mmu set paddr=00000000 count=1
D (97) boot: mapped partition table 0x8000 at 0x3f408000
D (104) boot: load partition table entry 0x3f408000
D (111) boot: type=1 subtype=2
I (115) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (125) boot: load partition table entry 0x3f408020
D (132) boot: type=1 subtype=1
I (136) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (146) boot: load partition table entry 0x3f408040
D (153) boot: type=0 subtype=0
I (158) boot:  2 factory          factory app      00 00 00010000 00100000
D (168) boot: load partition table entry 0x3f408060
D (174) boot: type=ff subtype=ff
I (179) boot: End of partition table
I (184) boot: Loading app partition at offset 00010000
D (191) esp_image: reading image header @ 0x10000
D (198) esp_image: reading 9 image segments
V (204) esp_image: loading segment header 0 at offset 0x10018
V (212) esp_image: segment data length 0xffe8 data starts 0x10020
D (221) bootloader_flash: mmu set paddr=00010000 count=2
V (296) esp_image: loading segment header 0 at offset 0x10018
V (297) esp_image: segment data length 0xffe8 data starts 0x10020
V (297) esp_image: loading segment header 1 at offset 0x20008
V (305) esp_image: segment data length 0x455c data starts 0x20010
D (314) bootloader_flash: mmu set paddr=00020000 count=1
V (340) esp_image: loading segment header 0 at offset 0x10018
V (340) esp_image: segment data length 0xffe8 data starts 0x10020
V (342) esp_image: loading segment header 1 at offset 0x20008
V (350) esp_image: segment data length 0x455c data starts 0x20010
V (359) esp_image: loading segment header 2 at offset 0x2456c
V (367) esp_image: segment data length 0x1a7c data starts 0x24574
D (376) bootloader_flash: mmu set paddr=00020000 count=1
V (390) esp_image: loading segment header 0 at offset 0x10018
V (392) esp_image: segment data length 0xffe8 data starts 0x10020
V (400) esp_image: loading segment header 1 at offset 0x20008
V (409) esp_image: segment data length 0x455c data starts 0x20010
[... 229 lines cut ...]
I (2789) GPIO: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:12
E (2799) GPIO: C:/msys32/esp-idf/components/driver/./gpio.c:143 (gpio_set_intr_type):GPIO interrupt type error
I (2809) GPIO: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:12
E (2819) GPIO: C:/msys32/esp-idf/components/driver/./gpio.c:143 (gpio_set_intr_type):GPIO interrupt type error
I (2829) GPIO: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:12
E (2839) GPIO: C:/msys32/esp-idf/components/driver/./gpio.c:143 (gpio_set_intr_type):GPIO interrupt type error
D (2849) camera: Resetting camera
D (2859) camera: Searching for camera address
E (2859) camera: Camera address not found
E (2859) camera_demo: Camera init failed with error = 131073
```

2. To be able to flash the code I have to power down the camera. Otherwise the following error is reported:

```
Krzysztof@tdk-kmb-op780 MSYS /esp/esp32-cam-demo
$ make flash
Flashing binaries to serial port com18 (app at offset 0x10000)...
esptool.py v2.0-dev
Connecting...

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
make: *** [C:/msys32/esp-idf/components/esptool_py/Makefile.projbuild:48: flash] Error 2
```
This error shows up no matter if I power up the board and camera using USB power supply or external power supply.

krzychb, 2016-11-21:
Ref: Layout of connections / using the camera port JP4 on ESP32_DevKitJ_V1

[image]

igrr, 2016-11-22:
I've seen the second issue — it's due to a pullup on the camera reset line. It is stronger than the internal pull-down on GPIO2 of the ESP32, so the chip can not go into programming mode. The fix for this is to connect GPIO2 to GND while flashing using a jumper wire (you may need to solder the second pin header for that).

Regarding LEDC, I haven't seen this issue. Will try with latest master.

It's probably not important yet at this point, but this camera uses D9 - D2 as its 8 data lines in 8-bit output mode. Pins D1 and D0 are only used in 10-bit mode. So you can ignore D1/D0 and plug the camera directly into the board.

krzychb, 2016-11-22:
Thank you for the tip regarding wiring.
I will fix it and then the picture above.

krzychb, 2016-11-23:
Regarding LEDC error, I understand there is some issue with setting up  XCLK signal on GPIO21, 
```
D (2759) camera: Enabling XCLK output
E (2759) LEDC: div param err,div_param=3
E (2759) camera: ledc_timer_config failed, rc=ffffffff
I (2769) LEDC: LEDC_PWM CHANNEL 0|GPIO 21|Duty 0004|Time 0
```
I can troubleshoot it using information provided in ESP32 Technical Reference Manual and LEDC API. I assume this signal should be @ 10MHz - https://github.com/igrr/esp32-cam-demo/blob/master/main/app_main.c#L54

This looks in line with OV7725 data sheet, that specifying CLK from 10 up to max 48 MHz. 

What do you think about it?

krzychb, 2016-11-23:
Still need to resolve issue with SSCB communication:

```
I (805) GPIO: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:12
E (805) GPIO: C:/msys32/esp-idf/components/driver/./gpio.c:143 (gpio_set_intr_type):GPIO interrupt type error
I (815) GPIO: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:12
E (825) GPIO: C:/msys32/esp-idf/components/driver/./gpio.c:143 (gpio_set_intr_type):GPIO interrupt type error
I (835) GPIO: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:12
E (845) GPIO: C:/msys32/esp-idf/components/driver/./gpio.c:143 (gpio_set_intr_type):GPIO interrupt type error
I (855) GPIO: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:12
E (865) GPIO: C:/msys32/esp-idf/components/driver/./gpio.c:143 (gpio_set_intr_type):GPIO interrupt type error
```

@igrr  any leads?

igrr, 2016-11-24:
I have pushed the fix for interrupt type error in 43752e5.
The original issue with `ledc_timer_config` failing is due to a bug in LEDC driver.
In ledc.c, line 148: 
```c++
    /*Fail ,because the div_num overflow or too small*/
    if(div_param <= 256 || div_param > LEDC_DIV_NUM_HSTIMER0_V) { //REF TICK
```
This should actually be `div_param < 256`. I'll put up a merge request in the ESP-IDF for this.

krzychb, 2016-11-24:
@igrr,

I did quick test and log looks promising.
More tests in the late afternoon.
Thank you for help!

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1a (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0008,len:0
load:0x3ffc0008,len:1580
load:0x40078000,len:3648
ho 0 tail 12 room 4
load:0x40080000,len:256
entry 0x40080034
I (348) heap_alloc_caps: Initializing heap allocator:
I (349) heap_alloc_caps: Region 19: 3FFB46AC len 0002B954 tag 0
I (349) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1
I (356) cpu_start: Pro cpu up.
I (360) cpu_start: Single core mode
I (365) cpu_start: Pro cpu start user code
rtc v134 Oct 20 2016 12:36:18
XTAL 40M
D (403) nvs: nvs_flash_init_custom start=9 count=6
D (438) phy_init: loading PHY init data from application binary
D (439) nvs: nvs_open phy 0
D (440) nvs: nvs_get cal_version 4
D (456) nvs: nvs_get_str_or_blob cal_mac
D (491) nvs: nvs_get_str_or_blob cal_data
D (533) nvs: nvs_close 1
I (722) phy: phy_version: 246, Nov 18 2016, 17:30:07, 0, 0

D (723) nvs: nvs_open phy 1
D (724) nvs: nvs_set cal_version 4 1
D (1116) nvs: nvs_set_blob cal_mac 6
D (1121) nvs: nvs_set_blob cal_data 1904
D (1183) nvs: nvs_close 2
I (1184) cpu_start: Starting scheduler on PRO CPU.
D (1185) camera: Enabling XCLK output
I (1185) LEDC: LEDC_PWM CHANNEL 0|GPIO 21|Duty 0002|Time 0
D (1185) camera: Initializing SSCB
I (1185) GPIO: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1195) GPIO: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1205) GPIO: GPIO[26]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1215) GPIO: GPIO[27]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
D (1225) camera: Resetting camera
D (1235) camera: Searching for camera address
D (1235) camera: Detected camera at address=0x21
D (1235) camera: Camera PID=0x77 VER=0x21 MIDL=0x7f MIDH=0xa2

D (1235) camera: Doing SW reset of sensor
D (1295) camera: Setting frame size
set_framesize: h=320 v=240
D (1325) camera: Allocating frame buffer (320x240, 76800 bytes)
D (1325) camera: Initializing I2S and DMA
I (1325) GPIO: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1325) GPIO: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1335) GPIO: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1345) GPIO: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1355) GPIO: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1365) GPIO: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1365) GPIO: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1375) GPIO: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1385) GPIO: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1395) GPIO: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1405) GPIO: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
D (1415) camera: Allocating DMA buffer #0, size=1280
D (1415) camera: Allocating DMA buffer #1, size=1280
D (1425) camera: Init done
D (1425) camera_demo: Starting camera capture
D (1425) camera: Waiting for VSYNC
D (1445) camera: Got VSYNC
D (1445) camera: Waiting for frame
D (1475) camera: Frame done
```

krzychb, 2016-11-25:
Both issues are resolved in master.

closed 2016-11-25 by krzychb

liubenyuan, 2016-11-28:
Can I config LEDC timer with freq=16MHz to 27MHz, as my mt9v034 only support xclk in this range.

krzychb, 2016-11-28:
Hi @liubenyuan,

Yes, you can configure LEDC timer with freq=16MHz.
In such case [LEDC channel duty depth](https://dl.espressif.com/doc/esp-idf/latest/api/ledc.html#_CPPv2N19ledc_timer_config_t7bit_numE) may be max 2.

Krzysztof

liubenyuan, 2016-12-01:
@krzychb Thanks! Now my mt9v034 works! I am waiting for your wifi-camera demo.

BTW, I extend `SCCB_Read` and `SCCB_Write` to `SCCB_Read16` and `SCCB_Write16`. However most of the codes are duplicated to the original one. Is this commit allowed to merge ?

igrr, 2016-12-01:
I would like to somehow get closer to the upstream OpenMV code, if possible. With the remote goal of running the whole OpenMV on the ESP32.

So you are welcome to make a PR which adds some notes to Readme.md that for mt9v034 support, 16-bit SCCB ops are required, and here's a link to the repository which implements these.

liubenyuan, 2016-12-01:
Well, how to make the `camera.c` compatible/extensible to different camera modules ? For example, mt9v034 do not have a `SCCB_Probe` function, it uses fixed slave address `0xB8` for write and `0xB9` for read ops.

liubenyuan, 2016-12-01:
And I think these codes in `camera.c` should be placed in application specific source files, i.e., `ov7725.c`

```c
    s_sensor.slv_addr = addr;
    s_sensor.id.PID  = SCCB_Read(s_sensor.slv_addr, REG_PID);
    s_sensor.id.VER  = SCCB_Read(s_sensor.slv_addr, REG_VER);
    s_sensor.id.MIDL = SCCB_Read(s_sensor.slv_addr, REG_MIDL);
    s_sensor.id.MIDH = SCCB_Read(s_sensor.slv_addr, REG_MIDH);
    ESP_LOGD(TAG, "Camera PID=0x%02x VER=0x%02x MIDL=0x%02x MIDH=0x%02x\n",
        s_sensor.id.PID, s_sensor.id.VER, s_sensor.id.MIDH, s_sensor.id.MIDL);

    /* debug mt9v034 */
    uint16_t chip_id = mt9v034_chip_id();
    printf("camera ID = %x\n", chip_id);

    ov7725_init(&s_sensor);

    ESP_LOGD(TAG, "Doing SW reset of sensor");
    sensor_reset();

    ESP_LOGD(TAG, "Setting frame size");
    framesize_t framesize = FRAMESIZE_QVGA;
    s_sensor.set_framesize(&s_sensor, framesize);
```

So in sensor `sensor_t` type struct, we should have two function calls, `camera_get_info` and `camera_set` to get and set some user specified values.