igrr/esp32-cam-demo #5: Why fix bit_num=3 and duty=4 in camera.c ?

liubenyuan, 2016-12-01:
These configuration will fail if I set xclk frequency to 16MHz.

krzychb, 2016-12-01:
Yes, it will fail because for 16MHz you can not get duty cycle precision of 3 bits.  You can get max. 2 bits.

In short xclk is obtained by dividing 80MHz internal clock. This clock can effectively count up to five (80MHz / 16MHz = 5). This limits precision of duty cycle. There is a picture in [Technical Reference Manual](http://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) in chapter "6.2.3 Channels" explaining how certain duty cycle is achieved.

Also, if you download latest esp-idf, besides failure message, you will also get explanation like: `requested frequency and bit depth can not be achieved, try reducing freq_hz or bit_num.`

liubenyuan, 2016-12-01:
Thanks, I will read that manual.

closed 2016-12-01 by liubenyuan