igrr/esp32-cam-demo #6: How does s_dma_desc get advanced?

cnlohr, 2016-12-01:
```cur_buffer``` seems to be updated, but, ```i2s_fill_buf``` seems not to use the buffer info.
```
    SET_PERI_REG_BITS(I2S_IN_LINK_REG(0), I2S_INLINK_ADDR, ((uint32_t) &s_dma_desc), I2S_INLINK_ADDR_S);
```
Is this to be expected?  Is there magic in the I2S subsystem to somehow select the right descriptor?

liubenyuan, 2016-12-01:
I am also stuck at this point. It seems like that they are using a ping pong buffer, but the parameter index is not used in `i2s_fill_buf`

I am using mt9v034 (which is a mono cmos sensor) and need to modify the dma size as well as the line_filter function.

cnlohr, 2016-12-02:
I'm trying to do this just to read data coming from the pins at a high rate, but I can't even get the IRQ to be called once.  This particular question though likely unrelated to my issue was the only one that really left me going ????

igrr, 2016-12-02:
Yes, it's probably a but. Thanks for pointing this out.

@cnlohr : if the interrupt doesn't fire it likely means that something is wrong with your input HSYNC & HREF & clock signals. Can you capture a scope trace of these signals?

cnlohr, 2016-12-02:
Well, specifically, I've disabled the camera portion - but I'm still driving the clock.  Perhaps I should go, re-enable the camera interface and and drive the HYSNC and HREF signals manually and see if the interrupts fire then.

igrr, 2016-12-02:
If I2S configuration has camera mode enable bit set, you need to make sure REF/SYNC signals are high. You can route them to a fixed high level using GPIO mux to avoid wasting a pin.

cnlohr, 2016-12-02:
I'm curious how do you do that?  I saw you doing this on this line:

```
gpio_matrix_in(0x38, I2S0I_H_SYNC_IDX, false);
```

Is there any map that shows all the functionality of the non-regular-gpio-pins in the matrix?

igrr, 2016-12-02:
This is mentioned in section 4.2.2 of the ESP32 Technical reference manual ("Peripheral Input via GPIO Matrix"), see Figure 7.
0x30 is the index which provides constant 0 input, 0x38 is the index which provides constant 1 input.

cnlohr, 2016-12-02:
[image]

cnlohr, 2016-12-03:
Got it working!  40 MSPS, so it's 1/2 the speed of the ESP8266's I2S bus from what I can tell... unless there's some clocking bit I need to set/reset... BUT... it can do that all 16-bits at a time, so the data pipe is pretty wide!!!!

The secret sauce was in the sync controls.

```
    gpio_matrix_in(0x38,    I2S0I_V_SYNC_IDX, false);
    gpio_matrix_in(0x38, 	I2S0I_H_SYNC_IDX, false);
    gpio_matrix_in(0x38,    I2S0I_H_ENABLE_IDX, false);
```

https://gist.github.com/cnlohr/2b9f8a26e891aa1a118f634cff9d04fe

cnlohr, 2016-12-03:
Errr.... I closed it because I solved a different problem.  Should I re-open to address first problem?

igrr, 2016-12-03:
Yep, let this be a reminder for me to fix the descriptors.

cnlohr, 2016-12-03:
Man... what I would give to get access to the magical tome of registers, instead of spending hundreds of hours blundering around in the dark till I find all the lightswitches I need to turn on. :-p.  Anyway, it would be good for all of the references to be as solid as possible so errors don't get propagated along.

igrr, 2016-12-03:
Actually there is no magical tome. What comments you see in header files are pretty much the same that we have. It takes time to describe everything and translate into English.

cranphin, 2016-12-05:
>  It takes time to describe everything and translate into English.

Can we have a Kickstarter for doing that for the ESP8266? :)

igrr, 2016-12-05:
@cranphin I can assure you, it's not one of the problems which can be easily solved just by throwing money at it :)
We'll do this for the ESP32 first, and then see if we have some time left for the ESP8266, before the next chip comes out.

cnlohr, 2016-12-05:
After playing with the ESP32... I don't know what next chip there could be.  The '32 already smashes just about every competitor into the dust.

cranphin, 2016-12-05:
ESP64 ofcourse :)

closed 2016-12-09 by igrr