igrr/esp32-cam-demo #9: Multi camera support

Oitzu, 2016-12-06:
Hoi,
i spent a couple of hours the last days trying to understand the code in order to reimplement a few of the original OpenMV features in https://github.com/Oitzu/esp32-camera.

Now, i'm kinda questioning myself if it wouldn't be better to modify the esp32-cam-demo to support multiple cameras and the gone camera settings.

Question: Would you approve such pull requests?

igrr, 2016-12-06:
I see this repository more as a simple demo, rather than a comprehensive solution for working with cameras on the ESP32. Well, maybe it's not simple enough, but still.

I think the best way of having multiple cameras supported would be to port OpenMV code to the ESP32. This may offer less instant gratification, but is surely better in the long term.

Initially we can start porting just the camera and image processing related parts of OpenMV. Micropython port is already available from Pycom, so we can reuse that. I think that other folks who frequent this repository would be interested in helping with OpenMV port as well. I don't have huge amount of free time, but I can lend hand with this as well.

Oitzu, 2016-12-06:
Okay, that clarifies my question. 
> Well, maybe it's not simple enough, but still.

It's great that we have this, without this we would probably still searching the whole next year for the right registers. :D I appreciate your help pretty much.

You are totally right, that porting the OpenMV code would be the best option.
Currently i'm concentrating on implenting the image processing related parts with keeping as much as possible from the OpenMV camera settings and registers.

After this is done, i could maybe concentrate to reimplement more of the OpenMV functions, if more people are interested in this.
But i'm still waiting for my camera (OV7620) for testing to be delivered. :/
At the moment i only have the datasheets to look at, but no actual hardware to test and probe it.

igrr, 2017-06-07:
Somehow we have drifted into actually supporting multiple (2) cameras in the same demo, although i think that porting OpenMV is still the way forward. Closing this for now.

closed 2017-06-07 by igrr