igrr/esp32-cam-demo #13: ov2640 can not output test pattern

donny681, 2016-12-07:
1.I'm trying to drive ov2640 and use openMV code(ov2640.c,ov2640.c,ov2640_reg.h),but I find that it  can not output test pattern.I set the framesize which is  QVGA.It should output 320x240 pixel,so I just need to change the code of camera init.I have not found the mistake of openMV code.Did you ever use Esp32 to drive ov2640,or I need to change i2s_fill_buf and line_filter_task.
------------------------------------------------------------------------------------------------------
D (1911) camera: Doing SW reset of sensor
D (1951) camera: Test pattern enabled
D (1951) camera: Setting frame size at 160x120
D (1981) camera: Allocating frame buffer (160x120, 19200 bytes)
D (1981) camera: Initializing I2S and DMA
I (1981) GPIO: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1981) GPIO: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1991) GPIO: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2001) GPIO: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2011) GPIO: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2021) GPIO: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2021) GPIO: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2031) GPIO: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2041) GPIO: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2051) GPIO: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2061) GPIO: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (2071) camera: Allocating DMA buffer #0, size=640
D (2071) camera: Allocating DMA buffer #1, size=640
D (2081) camera: Init done
D (2081) camera: Waiting for VSYNC
D (2141) camera: Got VSYNC
D (2141) camera: Waiting for frame
D (2201) camera: Frame done
D (2201) camera_demo: Done
                                                                                
                       =                                        =               
                                 =                                        =     
  =.......                 .....................         ....                   
...        .:-:.       ==...................  .     .:-:.       ==.............:
.......:::::::.   .....:::.       ==:..........::::::::::...    ..    ..........
..:-======-.   :..::::.....   =--..........-=======-.   ::.::::......  =--:.....
+--:.........:===+++++.  .::::..:--::::.=:--:.........:===+++++:  ..:::..:---:::
......-=====--+.::..... ....::-+++++++:.......:-=====--+..:........:.:::=++++++-
::::--=++==-:.......-========+...... ..::::::---+===-:.......-========+...... ..
   ::--::::------:.....  ..:=====--++....  .:---::::-----::..... ....-====--++..
..... .:-===-::::-::......  ..     .:---+...   .:-====:::::::.....   ..      .:-
...     .:--:       ==.   ..:-=====-:.......     .:-=-.      ==    ..:======-:..
::-=----:....        ..:.....::.  =    .::------::...        ...:::....::. =   .
...::------:...=   ..:::::::...          ..:--------::..=   ..::::::::::........
........           ..:--==---------++  .........            ..:--=----------++  
.::::---------::=   ......              .::------------::=  .......             
        ..:----::-::::::::+                     ...::---:::::::::::+            
                                    =                                        =  
                  =                                        =                    
                            =                                        =          
                                      =                                        =
                  ==                                       ==                   
                            ==                                       ==         
                                      ==                                       =
                   =                                        =                   
                             =                                        =         
                                       =                                        
                    +                                        =                  
                              =                                        =

krzychb, 2016-12-07:
Hi @donny681,

I have not tested ov2640. 
Judging by regularity / repeated characters, the problem may be due to noisy signal lines, 

Could you edit and update your post to [preserve formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/#fenced-code-blocks) of provided log?

Oitzu, 2016-12-13:
@donny681 maybe you can help me.
I'm also trying the ov2640 with following fork: https://github.com/Oitzu/esp32-cam-demo/tree/multi-cam-support

But i'm stuck in the SCCB communication. I get correctly back the slave-address (0x30), but failing at getting pid, ver etc... (all 0x00)

Oitzu, 2016-12-13:
Upadte: Only seems to happen after soft-reset of the esp. I didn't saw that because i missed the first boot.

Oitzu, 2016-12-14:
That moment when you try to invert vsync and fuck everything up....

Oitzu, 2016-12-14:
Thats what i got:
```
+@@@@@*####*@@@@@#####+++++..=..+++++.=..=======...===.===..:@@@@@@@@@@++@+++++@
@@@@@#####*@@@@@#####+++++=....+++++...=......+............:@@@@@@@@@@++@++++@++
@@@@@*####*@@@@@#####+++++....=+++++.......=..............+:@@@@@@@@@@+@@+++++++
@@@@@*####*@@@@@#####+++++..==.+++++...............=....=..:@@@@@@@@@@++@++++@++
@@@@@ ####*@@@@@#####+++++....=+++++........====.......=...:@@@@@@@@@@+++@@+++++
@@@@@*####*@@@@@#####+++++..=..+++++....======......=......:@@@@@@@@@@+++++@+++#
@@@@@*####*@@@@@###*#+++++.=.=.+++++................+......:@@@@@@@@@@+++@++++++
@@@@@ ####*@@@@@#####+++++..=..+++++.....==............=...:@@@@@@@@@@++++++++++
@@@@@#####*@@@@@#####+++++.....+++++......=....=.=..=.....=:@@@@@@@@@@++++++@@++
@=@@@@@  +  =====. +  ===................=...:@@@@@@@@@@+++++++@@+@@@@@*####*@@@
+========.========================  ++   +  @=====@@@@@=====@@@@@     ==+==.    
=++=========%@@+++@@@@#=====#####-====*    .@@@@#=====@@@@@=+.===========.=====.
++#@@@@@#####@@@@@#####+++++.....+++++........................@@@@@@@@@@++++++++
+@@@@@ ####*@@@@@####*+++++..==.+++++..................=....:@@@@@@@@@@+++++@+++
+@@@@@ ####*@@@@@####*+++++=..=.+++++....     =....=..=..=...@@@@@@@@@@++++++@@+
+@@@@@#####*@@@@@#####+++++.....+++++...=..........+.+......:@@@@@@@@@@+@+++++++
+@@@@@#####*@@@@@#####+++++.....+++++.====.===..=.........=.:@@@@@@@@@@+++++++++
+@@@@@*####*@@@@@#####+++++=..=.+++++....=.=====...=........:@@@@@@@@@@@+++++@++
+@@@@@*####*@@@@@#####+++++.=..=+++++............=.........=:@@@@@@@@@@++@@++++@
+@@@@@*####*@@@@@#####+++++...=.+++++............==.=.......:@@@@@@@@@@++++++++@
+@@@@@*####*@@@@@#####+++++.....+++++..................=....:@@@@@@@@@@+++++@+++
+@@@@@*####*@@@@@#####+++++.....+++++......=.======..==.....:@@@@@@@@@@++++@@+++
+@@@@@@###@@@@@@@@@@@*+ ++=.....+++++........=+.............@@@@@@@@@@*+++++++++
+@@@@@@####@@@@@@@@@#*+ ++=....+++++ ........======.==.=...=@@@@@@@@@@*+++++++++
+@@@@@@####@@@@@@@@@#*+ + =.=.==++ ++.=.=..=...=.=====..=...=@@@@@@@@@*+++++++++
+@@@@@@@@##@@@@@@####*++++=.....     ..=.==+.....=+.........=@@@@@@@@@*++++++++ 
+@@@@@@@@@@@@@@@@#@##*++++=...=.   ++......=========..+=..==@@@@@@@@@@* ++++++++
+@@@@@@@@@@@@@@@@###@*++++=..+=.+ +++...........====..=..===@@@@@@@@@@* ++++ +++
+@@@@@@@@@@@@@@@@@@@@*+++ =....=+++++................======.@@@@@@@@@@* +++ ++++
+@@@@@ ####*@@@@@####*+++++..==.+++++..=...............===.=:@@@@@@@@@@+++++++++
```

krzychb, 2016-12-14:
@Oitzu, this is fantastic!
I think you are just a tiny step from getting it clean.

Oitzu, 2016-12-14:
VSYNC and HREF look a bit noisy at the osci, but at least they look like they have the right pattern.
[image]

Oitzu, 2016-12-15:
At least my test picture looks like i'm doing something right:
[image]
@krzychb totally stealing your test picture. ;)

donny681, 2016-12-16:
Great,the photo is token by ov2640?It saves as JPEG format or others?

donny681, 2016-12-16:
Great,the photo is token by ov2640?It  output JPEG format or others?

donny681, 2016-12-16:
<pre><code> But i'm stuck in the SCCB communication. I get correctly back the slave-address (0x30), but failing at getting pid, ver etc... (all 0x00)</pre></code>
@Oitzu Before you read de address,you,must be software reset the camera,or out can't read the address.I don't know why set the GPIO2 down can't reset the camera,maybe it's  no use.

<pre><code>esp_err_t camera_init(const camera_config_t* config)
{
  ......
    ESP_LOGD(TAG, "Initializing SSCB");
    SCCB_Init(s_config.pin_sscb_sda, s_config.pin_sscb_scl);
    ESP_LOGD(TAG, "Resetting camera");
	
	gpio_config_t conf = {0};
    conf.pin_bit_mask = 1LL << s_config.pin_reset;
	conf.mode = GPIO_MODE_OUTPUT;
	
	gpio_config(&conf);
	/*software reset*/
	 SCCB_Write(0x30, 0xFF, 0x01);
	    SCCB_Write(0x30, 0x12, 0x80);

	    delay(10);
    gpio_set_level(s_config.pin_reset, 0);
    delay(1);
    gpio_set_level(s_config.pin_reset, 1);
    delay(10);
    ESP_LOGD(TAG, "Searching for camera address"); 
.....
}</pre></code>

donny681, 2016-12-16:
<pre><code>But i'm stuck in the SCCB communication. I get correctly back the slave-address (0x30), but failing at getting pid, ver etc... (all 0x00)</pre></code>
@Oitzu Before you read de address,you,must software reset the camera,or you can't read the address.I don't know why set the GPIO2 down can't reset the camera,maybe it's no use.

<pre><code>esp_err_t camera_init(const camera_config_t* config)
{
  ......
    ESP_LOGD(TAG, "Initializing SSCB");
    SCCB_Init(s_config.pin_sscb_sda, s_config.pin_sscb_scl);
    ESP_LOGD(TAG, "Resetting camera");
	
	gpio_config_t conf = {0};
    conf.pin_bit_mask = 1LL << s_config.pin_reset;
	conf.mode = GPIO_MODE_OUTPUT;
	
	gpio_config(&conf);
	/*software reset*/
	 SCCB_Write(0x30, 0xFF, 0x01);
	    SCCB_Write(0x30, 0x12, 0x80);

	    delay(10);
    gpio_set_level(s_config.pin_reset, 0);
    delay(1);
    gpio_set_level(s_config.pin_reset, 1);
    delay(10);
    ESP_LOGD(TAG, "Searching for camera address"); 
.....
}</pre></code>

donny681, 2016-12-16:
@Oitzu I try your code ,I find that the camera doesn't work.

Oitzu, 2016-12-16:
@donny681 
> Great,the photo is token by ov2640?It output JPEG format or others?

It's from the ov2640 in QVGA (windowed from uxga) 8-BIT YUV extracted from 422-YUV.

> Before you read the address,you,must software reset the camera,or you can't read the ov2640 address.I don't know why set the GPIO2 down can't reset the camera,maybe it's no use.

I adressed this issue in #18.

> @Oitzu I try your code ,I find that the camera doesn't work.

Any specific error? It's also not the latest code from yesterday. I need to clean up first.

Oitzu, 2016-12-16:
@donny681 just pushed the latest changes. Make sure to use the multi-cam-support branch, and configure it (menuconfig) the only include the ov2640 (i still need to edit the ov7725).

Mind that the standard app only outputs the ascii conversion, if you want the raw data you should output the fb (camera_get_fb()).

donny681, 2016-12-16:
@Oitzu  find that it's no use to read address after your new code and must power up again. Giving up for today.
Next step is to read JPEG data and  find that it can't allocate a large DMA buffer every time.

donny681, 2016-12-16:
Should I waiting for filling the first dma buffer and then go on waiting for filling the second buffer.When a small  DMA buffer is overflow, deal with the data and clean it up.

Oitzu, 2016-12-16:
> @Oitzu find that it's no use to read address after your new code and must power up again. Giving up for today.
> Next step is to read JPEG data and find that it can't allocate a large DMA buffer every time.

Maybe we figure it finally out in #18.
You can't allocate big DMA buffers? Whats the error? How big do you try to allocate?

> Should I waiting for filling the first dma buffer and then go on waiting for filling the second buffer.When a small DMA buffer is overflow, deal with the data and clean it up.

Is that a question? :D Thats the way how the code works at the moment, expect that the buffer should not overflow but just fill up.
The switch/flushing of the buffer happens in i2s_isr that is triggered by a full buffer.
The configuration of the "trigger" is here:
```
I2S0.rx_eof_num = s_buf_line_width;
```

Oitzu, 2016-12-18:
About the jitter: The alternative pinout from #18 made a huge difference in matter of noise.
Before:
[image]
After:
[image]

@krzychb: I'm making a note here: Huge success. :+1:

krzychb, 2016-12-18:
@Oitzu, always love to hear from you :smile: 
With alternate wiring and by preparing a daughter board as below  I was able to cut original wire length by two and get more rigid test jig.

Oitzu, 2016-12-18:
@krzychb this looks quite neat. :)

Playing arround this evening (without any good results yet) (I just don't understand enough of the platform yet to use it correctly) i accidently activated my test pattern.
Note: On the ov7620 the test pattern mixes up with the actual picture and produces strange vertical lines.
If a man don't know this, a man might spend hours searching what is going wrong.

igrr, 2017-06-07:
Closing this as OV2640 support (including the test pattern) is now in master.

closed 2017-06-07 by igrr