igrr/esp32-cam-demo #15: JPEG mode

Oitzu, 2016-12-08:
Forked from #11 to not go off-topic.

>  I also need to see the waveforms for JPEG case and check with hardware folks if this is something that I2S can support (although this is not something directly related to this specific issue).

I will do this, as soon as i get the cam. (Feels like taking forever)

Oitzu, 2016-12-08:
According to this code: https://github.com/iqyx/ov2640-stm32/blob/master/F4discovery/main.c
It SHOULD btw. look like this:
[image]

Oitzu, 2016-12-08:
Well its not completly clear from the code how often href rises and falls. Just that VSYNC goes high on the beginning of the frame and low at the end and that an active href marks valid data on the databus.

Oitzu, 2016-12-12:
[image]

This is more accurate. In the end only the dma and fb buffers needs to extended to a safety margin, and the transfer needs to be stoped at falling vsync.

Oitzu, 2016-12-15:
@igrr last one is the case. Can we get an example on how to work with an interrupt on vsync?

igrr, 2016-12-16:
Is there a known max length of data sent per HREF high period? Is there some guaranteed alignment, i.e. total size is dividable by 2 or by 4? Getting an interrupt is easy — check the gpio driver, it has some function to install interrupt handler. But i don't know how to tell the DMA FSM to flush all the data and end the transfer, especially if the FIFO contains an odd number of bytes. I'll try to clarify this.

Oitzu, 2016-12-16:
> Is there a known max length of data sent per HREF high period?
It's unfourtnatly not documented. Other implementations just streams in a big dma-buffer and flush it when vsync is going low.
Afterwards bytes are cut before and trailing the JPEG-Data, because this method seems kinda unclean but used widely. 

>  Is there some guaranteed alignment, i.e. total size is dividable by 2 or by 4?
Also not documented. :((

I will try to experiment a little bit arround.

Oitzu, 2016-12-22:
@igrr looking at the data and the osci, the data suggests that PCLK per HREF-HIGH stays at resolution width.
The number of HREF-HIGHS per VSYNC-HIGH seems to change with the compression rate.

So the dma-buffers could actually stay the same. The framebuffer should be set to maximum image size and trimed after capture.

Oitzu, 2016-12-24:
Hm.... doesn't seem to quite work out if i try it.
Maybe a big dma buffer and flushing on falling vsync would be better.

igrr, 2016-12-30:
Got myself an ov2640; will try to play with it on holidays.

Oitzu, 2016-12-30:
Thats great news. :)
I played arround also the last few days, at low PCLK-Rate i get a few bytes out of it that suggest JPEG-Data ("ff d8 ff e0 00 10 4a 46 49 46" jpeg-header), but get corrupted data or task-handle-errors afterwards.
I think my "PCLK per HREF-HIGH" theory was wrong, unfourtnatly my oscilloscope doesn't allow a precise measurment of that.
Maybe i need to stock up my gear and also get jtag-debug-hardware (only swd laying arround).

igrr, 2017-01-20:
@Oitzu I had a few minutes today to check with the folks who had designed I2S peripheral, and there seems to be a way to get all the data out of I2S even when RXEOF interrupt doesn't fire. Will try using your OV2640 code.

Oitzu, 2017-01-20:
@igrr great to hear that you found some time. :)
I pushed some of my jpeg-tests to https://github.com/Oitzu/esp32-cam-demo/tree/ov2640-jpeg-playground
It's absolutely messy, and probably completly on the wrong path but well... :/
Mind: It uses wifi+httpd to output the raw-data and uses the alternative pinout.

igrr, 2017-01-20:
Yep, I have to find more time for this — apparently there are people who actually want to use JPEG sensors with the ESP32 for mass production, and we have to support them. Thanks for the code and the pointers.

Oitzu, 2017-01-20:
Well that is great to hear, for me. :P
I'm actually validating to use the esp32 on a project.  But only on the laughable scale of 10-20 units. (for the start)

Any open information about the guys that want to mass produce?

igrr, 2017-01-22:
Well, based on your code i was able to get mjpeg streaming, a few frames per second :) And thanks for dropping a web server in there — it has really helped debugging!

I need make this thing at least slightly faster and do some, will push my work-in-progress branch once I'm done for the day.

Oitzu, 2017-01-22:
Wow thats great, and is exactly what i wanted to accomplish, but stopped after not getting one complete frame. :)

Note: If you want push in reference to my "quick and dirty"-branch mind that i rewrote history of the commit 2 days ago. I accidently dropped wlan credentials in it (that i already changed on my side). I should have implemented them in the sdkconfig at the first place. *roll-eyes*

I'm drilled to see how it is implemented.

Oitzu, 2017-01-28:
@igrr any news on this? :)

igrr, 2017-02-07:
The workaround for #20 has been found, and we can now get JPEG stream at 20MHz pixel clock. Need to think how to integrate this properly though, as clock rates <=10MHz and >10MHz require different DMA setups.

Oitzu, 2017-02-08:
Maybe a getter method implemented into the camera-specific files that calculates the expected pclk based on xlck/multipliers/dividers, resulting on different function calls on dma init?

igrr, 2017-02-09:
Pushed my work-in-progress branch. At the moment I have likely broken OV7725 (i.e. non-jpeg) support, will fix it a bit later.
OV2640 works in jpeg mode both at 10MHz and 20MHz. Server can now do mjpeg streaming (although the frame rate leaves a lot to be desired).

Oitzu, 2017-02-09:
Great. :+1: Quite a lot changes. Need to play arround with it the next few days.
Did you ever tried to bump it up to UXGA?

Oitzu, 2017-02-09:
Cloned it. Modified the pin configuration back.
Getting an error while the DMA buffers are allocating.
https://gist.github.com/Oitzu/842b5027c59e3ed435f8f82f697bf77e

I updated inbefore my toolchain also.

igrr, 2017-02-09:
Can you please try with log level set to "verbose" in menuconfig? Also you may try using xtensa-esp32-elf-addr2line on the addresses in backtrace line to see where exactly it is crashing.

Oitzu, 2017-02-09:
```

Oitzu, 2017-02-09:
Verbose output here: https://gist.github.com/Oitzu/e6c14089fadf4dc488a4c71b806d0c42

Strangly enough the backtrace differs?

H-LK, 2017-02-10:
Getting an error during i2c check of camera:
D (1843) camera: Detected camera at address=0x30
D (1853) camera: Camera PID=0x00 VER=0x00 MIDL=0x00 MIDH=0x00

Cables are very short and testet.

Sparkfun ESP32 THING + WaveShare OV2640

Oitzu, 2017-02-10:
@H-LK :
Well obviously first check your cables first against the config and a pinout of your board.
Especially the vcc/gnd/SIDC/SIOD connections.

Second: Try a "cold boot". It happens that the ov2640 doesn't reset correctly on soft reboots.
1. Disconnect USB.
2. Press and hold Reset on the esp32.
3. Connect USB and wait till the usb-serial chip is completly connected.
4. Open Serial Terminal.
5. Release Reset.
With this way you get the first serial output from your esp32 on "cold boot".

H-LK, 2017-02-10:
@Oitzu works fine :)
I am getting the same type of error like Oitzu:
https://gist.github.com/H-LK/7bfcc058d079d4bf058580bdb45f1f57

Oitzu, 2017-02-10:
Well the error goes away if i activate the multi-core freertos support.
I don't know if this is the right way todo. I don't fully understand the differences between uni and multicore freertos and the default sdkconfig says otherwise: https://github.com/igrr/esp32-cam-demo/blob/feature/ov2640/sdkconfig#L131

Also i do not get pictures out of it, which may have something to do with this warnings:
W (27866) camera: queue send failed (0), dma_received_count=20

igrr, 2017-02-10:
I had dual core mode enabled, but probably forgot to commit sdkconfig changes.

The queue send error indicates that camera is pushing data out faster than the line filter task process it. Try turning down debug level... I think it worked for me with "debug" level at 20MHz camera clock, but maybe there's something else that I'm missing. Alternatively, try reducing the camera clock to 10MHz.

Oitzu, 2017-02-10:
I had to reduce the clock down to 5MHz to get a not gibberish picture, but that works! :)

I probably need to work on the hardware connection between camera and esp32 to get better/faster results.

igrr, 2017-02-10:
Yes, one thing with jpegs is that if you get a single byte wrong, the the rest of the picture will be garbage. I'm using the ESP-WROVER-KIT here, it works okay up to 20MHz.

In the next chip we'll bump max i2s slave clock to 40MHz, to support those higher resolution modes.

Oitzu, 2017-02-10:
So the higher resolution modes are currently not implemented because that would be painfully slow?

With "the next chip" you are speaking of the next revision of the esp32?
I hope that also someone will produce a module with a ipex connector on the next revision.

igrr, 2017-02-10:
Regarding higher resolutions (uxga?) — I just used the code from openmv, so I suppose it's just a matter of adding the right register configuration in ov2640.c. I have tested SVGA (800x600) which worked fine, but that's the highest resolution which I personally need.

Regarding the next chip, I was referring to the single-core version of the esp32. The next silicon revision of dual-core esp32 doesn't have these changes. The new module based on the ESP32, which has 4MB pSRAM, also comes with ipex connector. PSRAM will come handy to store larger frame buffers as well.

Oitzu, 2017-02-11:
Will definitly try the higher resolution. In the meantime i pushed a few smaller enhancements. (No more fighting about the pin configuration)

Didn't know that there is a single-core ESP32 in the queue. Is there anywhere detailed information about the differences between different versions and revisions?

Oitzu, 2017-02-11:
Was investigating the higher resolution and was baffled about the buffer allocation.
We are actually allocating more buffer than we need, or?
Also we do send a lot of junk after the picture? (Should end with 0xff 0xd9)

igrr, 2017-02-11:
I'm allocating (fb_size)/compression_ratio, where compression ratio is estimated based on jpeg quality parameter. That's not perfect, if you have any suggestion please let me know.

Regarding junk — quite probably. I think the size is rounded up to the end of the line.

Oitzu, 2017-02-18:
> That's not perfect, if you have any suggestion please let me know.

At the moment, no. I was just curious from where this number is coming from.
Hard to say what the maximum framesize could be, without knowing of the internal workings of the ov2640 jpeg compression.

Possibly there is a method like: "header + (number_of_8x8_chunks * max_chunk_size / quality_factor) + footer (+junk)" but i'm not quite sure about the max chunk size and the quality factor.

Maybe i just wait for the new ESP32 module and all my space restrictions will be gone. ;)

radiumray, 2017-05-08:
@Oitzu  i am trying https://github.com/Oitzu/esp32-cam-demo/tree/ov2640-jpeg-playground
but i got nothing in my broswer, 
i can get data ascii data on serial for another branch of git code

Oitzu, 2017-05-08:
@radiumray could you give us the serial output? Maybe something in there gives a clue what is going wrong.

igrr, 2017-05-08:
There's also [feature/ov2640](https://github.com/igrr/esp32-cam-demo/tree/feature/ov2640) branch in this repo, which i need to merge one day. I think the only thing not finished there are non-JPEG modes (which are broken).

radiumray, 2017-05-08:
@Oitzu i was useing OV7725, there is no serial output for jpeg-playground branch
i have change camera config to this:
    camera_config_t config = {
        .ledc_channel = LEDC_CHANNEL_0,
        .ledc_timer = LEDC_TIMER_0,
        .pin_d0 = 4,
        .pin_d1 = 5,
        .pin_d2 = 18,
        .pin_d3 = 19,
        .pin_d4 = 36,
        .pin_d5 = 39,
        .pin_d6 = 34,
        .pin_d7 = 35,
        .pin_xclk = 21,
        .pin_pclk = 22,
        .pin_vsync = 25,
        .pin_href = 23,
        .pin_sscb_sda = 26,
        .pin_sscb_scl = 27,
        .pin_reset = 2,
        .xclk_freq_hz = 10000000,
        .pixel_format = CAMERA_PF_JPEG,
        .frame_size = CAMERA_FS_VGA,
        .jpeg_quality = 15

Oitzu, 2017-05-08:
@radiumray not even "Enabling XCLK output" -> "Initializing SSCB" etc...?
Could you increase the debug output level?
Absolutly no serial output would be very strange and would be a hint to a more basic problem.

@igrr The jpeg-playground branch is basicly the same just added more configuration-options and fixes/adds direct jpeg display in browser.

radiumray, 2017-05-09:
hi @Oitzu  thanks a lots for your advise, (it is seem like not supported OV7725, Is not it?)
i have already increase my debug level,below is my serial output:
}D (473) camera: Test pattern enabled[0m
[0;31mE (473) camera: JPEG format is only supported for ov2640[0m
C:/esp/esp-idf/components/freertos/heap_regions.c:369 (vPortFreeTagged)- assert failed!
abort() was called at PC 0x40084ef8 on core 0
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40008155:0x3ffb8760 0x40007d16:0x3ffb8780 0x40087320:0x3ffb87a0 0x40081c6c:0x3ffb87c0 0x4000beca:0x3ffb87e0 0x400dd41a:0x3ffb8800 0x400dbdf3:0x3ffb8830 0x400d0ba1:0x3ffb88c0

================= CORE DUMP START =================
LBMAAAYAAAB8AQAA
UPT6P6CG+z9Mifs/
sIX7P+CI+z8UAAAACC/7Pwgv+z9Q9Po/AC/7PxgAAACU5fo/lOX6P1D0+j8AAAAA
AQAAAFB5+z9tYWluAApDnxYZc/RO6pAAAAAAAEyJ+z8AAAAAIQAGAAEAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWOP6P8Dj+j8o5Po/
AAAAAAAAAAABAAAAAAAAADAeQD8AAAAASB0AQAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADxV64=
g2gIgFWBAEA2CQYAFn0AgGCH+z9YAAAAWAAAABAAAAD/////AAAAAPD///+xLfs/
AQAAAC4AAAAuAAAADAAAAP////8AAAAA/v///wUAAAABAAAAAAAAAP0UAEANFQBA
/v///wwAAAD/////HBQIQAAAAAAAAAAAAAAAAAAAAAAAAD+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAgAAAAIAAAA1Ib7P/////8gcwiAgIf7P/Kl8CvqpfAr
EAAAAP////8AAAAA8P///2wcCICgh/s/8qXwK8Dj+j+khvs/AAAAAAYAAAAAAAD/
yr4AgMCH+z/Q9Po/8qXwKwEAAACAhPs/gIT7PwIAAAAa1A2A4If7P/Kl8CtkHAhA
wIf7PwwAAAAAAAAABQAAAPO9DYAAiPs/5J77PwYBAADsLEA/HwAAAAEAAAAFAAAA
oQsNgDCI+z+ASAAAAAAAAOABAAAwiPs/gEgAAAAAAAB3AAAAAwAAAAoAAAAKAAAA
AAAAAMCI+z8AAAAAAAAAAOCI+z8AAAAAAAAAAAAAAAACAAAAFQAAABoAAAAbAAAA
IwAAACIAAAAnAAAAJAAAABMAAAASAAAABQAAAAQAAAAZAAAAFwAAABYAAACAlpgA
AAAAAAAAAAADAAAACgAAAA8AAAAAAAAAAAAAAAAAAAAjAAYAUPT6PwAAAAABAAAA
AAAAAOCI+z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAADsiPs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
fP36PzCN+z+Ujvs/
MI37PzCO+z+kfrgR9C77PwD8+j98/fo/7C77PxkAAADNaVaFELP8dHz9+j8AAAAA
AAAAAFiJ+z9JRExFAIgr7h8D6HdMsm8AAQAAAJSO+z8AAAAAIA0GAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWOP6P8Dj+j8o5Po/
AAAAAAAAAAABAAAAAAAAADAeQD8AAAAASB0AQAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABZqDc=
EBUIQAQODUAwDQYA4EMIgPCN+z8IAAAAAQAAACMABgD4+/o/AAAAAAEAAAAULvs/
0I37PwAAAAAgDQYAIwAGAFDx+j8DAAAAIwoGAAAAAAABAD+zAAAAAAAAAAAAAAAA
AAAAAA0XCEBQ8fo/jFsIQAAAAAAAAAAAAAAAAAAAAAABAD+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAACMABgBQ8fo/AwAAACMKBgAAAAAAEI77PwAAAAAAAAAA
IwAGAPj7+j8AAAAAAQAAAAAAAAAwjvs/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPI77PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAA=
+Pv6P4D6+j/s+/o/
gPr6P4D7+j/jV9kRhP36P/Qu+z/4+/o/7C77PxkAAAB74DGR+8LnRPj7+j8AAAAA
AAAAALD2+j9JRExFAPBHXU4t1sjDouYAAAAAAOz7+j8AAAAAIQAGAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWOP6P8Dj+j8o5Po/
AAAAAAAAAAABAAAAAAAAADAeQD8AAAAASB0AQAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABsSoQ=
EBUIQAQODUAwAQYA4EMIgED7+j8IAAAAAQAAACMABgD4+/o/AQAAAAEAAAAULvs/
IPv6PwAAAAAgAQYAIwAGAICG+z8DAAAAIwAGAAAAAACA+/o/AAAAAAAAAAAAAAAA
AAAAAA0XCECAhvs/jFsIQAAAAAAAAAAAAAAAAAAAAAAAAD+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAYPv6PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAACA+/o/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAjPv6PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
[... 30 lines cut ...]
AAAAAAMAAAAAAAAAAAAAAP////8AAAAAAAAAAAAAAAAk7fo/AAAAAAEAAAAAAAAA
AAAAAADy+j8AAAAAAAAAAAEAAAAAAAAAAAAAAAAAAADcDgiAYH3+PygAAAAoAAAA
AAAAAAAAAAAM8vo/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
WOv6P8Dp+j9M6/o/
wOn6P+Dq+j+BLr/fgPL6P3Au+z9Y6/o/aC77PwEAAADo5fo/6OX6P1jr+j/g5fo/
GAAAABDm+j9pcGMwAPudsGam6/sk26gAAAAAAEzr+j8AAAAAIQAGABgAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWOP6P8Dj+j8o5Po/
AAAAAAAAAAABAAAAAAAAADAeQD8AAAAASB0AQAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACl4og=
EBUIQLdYCEAwAAYAuxAIgIDq+j+85fo/AAAAAATm+j8AAAAAAQAAAAEAAACFWQiA
YOr6P9wA8D8BAAAA4DD7P0A7/j9kE/s/AgAAAAAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAAA0XCEBAO/4/jFsIQAAAAAAAAAAAAAAAAAAAAAAAAD+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAwOr6PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAP////8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAODq+j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAjDgiA8Dv+P1gu+z/EMfs/
AAAAAAAAAADs6vo/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
================= CORE DUMP END =================
CPU halted.

Oitzu, 2017-05-09:
@radiumray yes the JPEG format is only available to the ov2640.
The mode relies on the internal jpeg processing capabilities of the ov2640. 
The OV7725 doesn't have this feature afaik.
Also we have no free memory, at the moment, to do some fancy yuv/rgb to jpeg compression.

radiumray, 2017-05-10:
@Oitzu  a huge thanks

radiumray, 2017-05-12:
hi @Oitzu i get OV2640 already, i still can't get image on my brower, my address is:
http://192.168.199.222/get
http://192.168.199.222/stream

the debug info below is look fine:
D (843) nvs: nvs_get_str_or_blob sta.apinfo[0m
D (843) nvs: nvs_get_str_or_blob ap.ssid[0m
D (843) nvs: nvs_get_str_or_blob ap.mac[0m
D (843) nvs: nvs_get_str_or_blob ap.passwd[0m
D (853) nvs: nvs_get_str_or_blob ap.pmk[0m
D (853) nvs: nvs_get ap.chan 1[0m
D (853) nvs: nvs_get ap.authmode 1[0m
D (863) nvs: nvs_get ap.hidden 1[0m
D (863) nvs: nvs_get ap.max.conn 1[0m
D (863) nvs: nvs_get bcn.interval 2[0m
D (873) nvs: nvs_get ap.phym 1[0m
D (873) nvs: nvs_get ap.phybw 1[0m
D (873) nvs: nvs_get ap.sndchan 1[0m
D (883) nvs: nvs_set_blob sta.mac 6[0m
D (883) nvs: nvs_set_blob ap.mac 6[0m
I (883) wifi: Init dynamic tx buffer num: 32
I (883) wifi: Init dynamic rx buffer num: 64
I (883) wifi: wifi driver task: 3ffd2740, prio:23, stack:3584
I (893) wifi: Init static rx buffer num: 10
I (893) wifi: Init dynamic rx buffer num: 64
I (903) wifi: Init rx ampdu len mblock:7
I (903) wifi: Init lldesc rx ampdu entry mblock:4
I (913) wifi: wifi power manager task: 0x3ffd7b04 prio: 21 stack: 2560
I (913) wifi: wifi timer task: 3ffd8ba0, prio:22, stack:3584
D (923) phy_init: loading PHY init data from application binary[0m
D (923) nvs: nvs_open phy 0[0m
D (933) nvs: nvs_get cal_version 4[0m
D (933) nvs: nvs_get_str_or_blob cal_mac[0m
D (933) nvs: nvs_get_str_or_blob cal_data[0m
D (943) nvs: nvs_close 3[0m
[0;32mI (973) phy: phy_version: 350, Mar 22 2017, 15:02:06, 0, 0[0m
I (973) wifi: mode : sta (24:0a:c4:04:03:00)
D (973) event: SYSTEM_EVENT_STA_START[0m
I (973) wifi: sleep disable
D (973) tcpip_adapter: check: local, if=0 fn=0x40113270
[0m
D (983) tcpip_adapter: call api in lwip: ret=0x0, give sem[0m
D (983) tcpip_adapter: check: remote, if=0 fn=0x40113270
[0m
I (1713) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (2373) wifi: state: init -> auth (b0)
I (2373) wifi: state: auth -> assoc (0)
I (2383) wifi: state: assoc -> run (10)
I (2633) wifi: connected with Microduino_IoT, channel 6
D (2633) event: SYSTEM_EVENT_STA_CONNECTED, ssid:Microduino_IoT, ssid_len:14, bssid:d4:ee:07:22:26:6c, channel:6, authmode:3[0m
D (2643) tcpip_adapter: check: local, if=0 fn=0x4011344c
[0m
D (2643) tcpip_adapter: call api in lwip: ret=0x0, give sem[0m
D (2653) tcpip_adapter: check: remote, if=0 fn=0x4011344c
[0m
D (2653) tcpip_adapter: check: local, if=0 fn=0x401137c8
[0m
D (2663) tcpip_adapter: dhcp client init ip/mask/gw to all-0[0m
D (2663) tcpip_adapter: dhcp client start successfully[0m
D (2673) tcpip_adapter: call api in lwip: ret=0x0, give sem[0m
D (2673) tcpip_adapter: check: remote, if=0 fn=0x401137c8
[0m
D (2713) event: SYSTEM_EVENT_STA_GOTIP, ip:192.168.199.222, mask:255.255.255.0, gw:192.168.199.1[0m
[0;32mI (2713) event: ip: 192.168.199.222, mask: 255.255.255.0, gw: 192.168.199.1[0m
[0;32mI (2723) camera_demo: Free heap: 155892
[0m
[0;32mI (2723) camera_demo: Camera demo ready[0m
[0;32mI (2733) camera_demo: open http://192.168.199.222/get for single frame[0m
[0;32mI (2743) camera_demo: open http://192.168.199.222/stream for multipart/x-mixed-replace stream (use with JPEGs)[0m
I (12383) wifi: pm start, type:0

D (19993) camera_demo: Image requested.[0m
D (19993) camera: Waiting for positive edge on VSYNC[0m
D (19993) camera: Got VSYNC[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20003) camera: gpio isr, cnt=0[0m
D (20113) camera: Waiting for frame[0m
D (20123) camera: gpio isr, cnt=0[0m
D (20123) camera: gpio isr, cnt=0[0m
D (20123) camera: gpio isr, cnt=0[0m
D (20123) camera: gpio isr, cnt=0[0m
D (20123) camera: gpio isr, cnt=0[0m
D (20123) camera: gpio isr, cnt=0[0m
D (20123) camera: gpio isr, cnt=0[0m

Oitzu, 2017-05-17:
Never seen this SCCB_Write errors.
Try following to debug: 
Press and hold reset on your esp32 board.
Plug in USB.
Open Serial connection.
Release the reset button.

This gets you the fresh startup output without any software resets or whatever in between.

radiumray, 2017-05-20:
@Oitzu hi
got another OV2640 from different factory, and it is almost work perfectly
my browser can reciver data but display error
i have try to print s_state->fb to Serial when http://192.168.199.222/get
my s_state->fb hex data is not to begin FF D8 ..........and end with FF D9, it is not like jpeg format

my s_state->fb hex data Serial print is like that:
ff 10 49 01 00 00 ff 43 0a 0b .....................................................................
..................................................................................................................................
...................................................................................................................................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 14 00 0d

Oitzu, 2017-05-20:
The end of the jpeg is currently not junk-free but should work neither the less.
Try to lower the clockrate also make sure to turn down the log-output-level to prevent timing issues.

JuRage, 2017-06-01:
can anyone help me with this error code, i am sorry i know yall are way beyond this but i am trying to get this code working.
D (240447) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (240447) tcpip_adapter: check: remote, if=0 fn=0x40113408
0x40113408: tcpip_adapter_down_api at C:/esp/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:801

hopkinskong, 2017-06-02:
@Oitzu @igrr Can you guys confirm you can't get proper JPEG image beyond 20MHz XCLK? I have bumped up the XCLK to 20MHz (confirmed with logic analyzer), and I start to have missing bytes (like JPEG header "JFIF" turns to "JFF").

btw, I can get like 8~10fps with MJPEG under 800x600 under 20MHz XCLK, I am using long wires though.

hopkinskong, 2017-06-02:
Also, when I increased my core speed to 240MHz (default 160MHz), I can't see the images anymore. Frames seems still arrives, but corrupted, any idea why?

Oitzu, 2017-06-06:
@hopkinskong regarding the 20MHz XCLK: I couldn't reach 20MHz successful yet.
But i have a rather messy test-setup for this, and it very much depends on cable length and connection quality.

hopkinskong, 2017-06-06:
@Oitzu Mine is messy too, I have 40CM long cables between the ESP32 and EVERY pins of my OV2640. With the logic analyzer, I can confirm that with >20MHz XCLK will in a result of >20MHz PCLK, which makes the I2S receiver missing some received bytes.

closed 2017-06-07 by igrr

anuprao, 2020-01-07:
Hi Everyone, I am trying to understand how the length of the JPG is assessed. I

commit 6d61c1d5 (2017-06-06): 20MHz and JPEG support

- Fixes #15 using special sampling mode which can capture the last
  sample before HSYNC goes low.
- Implements #20
- Breaks grayscale mode with OV7725