igrr/esp32-cam-demo #16: Missing esp_intr_alloc.h

krzychb, 2016-12-09:
With commit https://github.com/igrr/esp32-cam-demo/commit/cf5d8a1637544ad2b9209c4faebd5c3735b9af3b I am getting the following error:

C:/msys32/esp/esp32-cam-demo/components/camera/camera.c:44:28: fatal error: esp_intr_alloc.h: No such file or directory
 #include "esp_intr_alloc.h"
                            ^
compilation terminated.

donny681, 2016-12-10:
I meet the same problem,Maybe he forget to post the code..

donny681, 2016-12-10:
I facethe same problem,Maybe he forget to post the code..

donny681, 2016-12-10:
I face the same problem,Maybe he forget to post the code.

krzychb, 2016-12-10:
I hope this file is stacked with broken master branch https://twitter.com/i_grr/status/807192758783836160

igrr, 2016-12-11:
Sorry about that. No, the master branch of ESP-IDF was broken, hence some commits (including the new interrupt allocation feature) haven't been deployed to Github.

krzychb, 2016-12-14:
For latest esp-idf (with commits as of 13-Dec-2015) this error is not reported anymore.

closed 2016-12-14 by krzychb