igrr/esp32-cam-demo #18: Pulling reset line low

Oitzu, 2016-12-15:
Could it be that the camera resetting at the beginning of the camera_init() is not working correctly?
I couldn't observe any gpio level chances while it should reset.

I chanced the code to:
```C
    gpio_config_t conf = {0};
    conf.pin_bit_mask = 1LL << s_config.pin_reset;
    conf.mode = GPIO_MODE_OUTPUT;
    gpio_config(&conf);
	
    gpio_pulldown_en(s_config.pin_reset);
    gpio_set_level(s_config.pin_reset, 0);
    delay(10);
    gpio_pulldown_dis(s_config.pin_reset);
    gpio_set_level(s_config.pin_reset, 1);
    delay(10);
```
and it works, but there are probably more clean ways to do this.

krzychb, 2016-12-15:
Hi @Oitzu,

Have you checked https://github.com/espressif/esp-idf/blob/master/examples/02_blink/main/blink.c  including comments inside the code?

What particular pin do you use for resetting the camera?

Oitzu, 2016-12-15:
@krzychb I'm using pin 2, as predefined.
Maybe there is a difference between the ov2640 and the ov7725?
The OV2640 states that reset is activly pulled up in the camera therefore needs to be pulled down to go into reset.

krzychb, 2016-12-15:
Sneaky traps everywhere :-)
[image]

Oitzu, 2016-12-15:
[image]

krzychb, 2016-12-15:
Regarding pinout and pullups I have changed wiring in order to:

- avoid issue with internal pullups - https://github.com/igrr/esp32-cam-demo/issues/2#issuecomment-262207930
- have clean, easy to connect wiring.

Alternate wiring for the above criteria is as follows:

[image]

```
.pin_d0 = 35,
.pin_d1 = 17,
.pin_d2 = 34,
.pin_d3 = 5,
.pin_d4 = 39,
.pin_d5 = 18,
.pin_d6 = 36,
.pin_d7 = 19,
.pin_xclk = 27,
.pin_pclk = 21,
.pin_vsync = 22,
.pin_href = 26,
.pin_sscb_sda = 25,
.pin_sscb_scl = 23,
.pin_reset = 15,
```

Oitzu, 2016-12-15:
Ah, thanks! :+1: 
Funny enough i never got a problem with flashing the esp32.

I try your layout tommorow, maybe this even helps to get the lines shorter and the noise lower.

closed 2017-06-07 by igrr