igrr/esp32-cam-demo #19: Enabling "Watchdog task" is causing application to get stuck on random basis waiting for a frame

krzychb, 2016-12-16:
Application gets stuck on the following step:
```
D (2825) camera: Waiting for frame
```

This happens on random basis, after reset and (always) successfully displaying the first frame.
Then, in two out of three cases, application gets stuck on the second frame.
If second frame is displayed successfully, then application continue to operate normally.

To replicate this issue use settings as on the left below. 
Settings on the right do not cause any issues and are as used so far in this repository.

[image]

To change settings go to `Component config > ESP32-specific config > Task watchdog`
 
I have discovered this issue when checking generation of default `sdkconfig` file with `make defconfig`.
By default "Wathdog task" is enabled.

krzychb, 2016-12-30:
For the record attached is `sdkconfig` that is not causing this issue. 
Remove .txt extension if you like to use it in your project.
[sdkconfig.txt](https://github.com/igrr/esp32-cam-demo/files/678673/sdkconfig.txt)

Now, if you delete it, as well as `sdkconfig.old` and run `make defconfig` to get default configuration (or `make menuconfig`), the application gets stuck as described above.

I see new development with current `esp-idf`. If you disable the Watchdog task, it will not resolve the issue. This indicates that some other settings in `sdkconfig` may be the troublemaker. So far I was unable to pin point the settings. Fortunately there are no many differences.

krzychb, 2016-12-30:
To complete this troubleshooting session, below is summary of differences between DEFAULT and GOOD configuration files. 

1. CONFIG_TASK_WDT
[image]

2. CONFIG_FREERTOS_UNICORE
3. CONFIG_FREERTOS_CHECK_STACKOVERFLOW_NONE
[image]

4. CONFIG_LOG_DEFAULT_LEVEL_INFO
[image]

Removing each individual difference in DEFAULT configuration does not resolve the issue.

igrr, 2017-06-07:
Fixed in #31

closed 2017-06-07 by igrr