igrr/esp32-cam-demo #20: After fixing Issue #11, OV7725 is unable to operate @ 20MHz XCLK

krzychb, 2016-12-18:
After increasing XCLK from 10 to 20MHz, output of test pattern of OV7725 looks as follows:

```
D (1693434) camera: Waiting for positive edge on VSYNC
D (1693444) camera: Got VSYNC
D (1693444) camera: Waiting for frame
D (1693454) camera: Frame done
D (1693454) camera_demo: Done
|@++++++++++          **********          @@@@@@@@@@----------@@@@@@@@@@+++++++++|
|@===-:::::::::          ++++++++++          **********          @@@@@@@@@@------|
|@+          **********          @@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@|
|:+++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::  |
|@*****          @@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%|
|+++++++++++**********                    @@@@@@@@@@@@@@@@@@@@+---------+++++++++|
|#@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########|
|+++++++++++*********=                   .@@@@@@@@@@@@@@@@@@@@---------=+++++++++|
|+---@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++======|
|@+          **********          @@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@|
|:+++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::  |
|@*****          @@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%|
|+++++++++++**********                    @@@@@@@@@@@@@@@@@@@@+---------+++++++++|
|#@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########|
|+++++++++++*********=                   .@@@@@@@@@@@@@@@@@@@@---------=+++++++++|
|+---@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++======|
|@+          **********          @@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@|
|:+++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::  |
|@*****          @@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%|
|+++++++++++**********                    @@@@@@@@@@@@@@@@@@@@+---------+++++++++|
|#@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########|
|+++++++++++*********=                   .@@@@@@@@@@@@@@@@@@@@---------=+++++++++|
|+---@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++======|
|@+          **********          @@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@|
|:+++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%**********++++++++++==========::::::::::  |
|@*****          @@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%|
|+++++++++++**********                    @@@@@@@@@@@@@@@@@@@@+---------+++++++++|
|*@@@@@@@@@----------@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%**********|
|+++++++++++*********=                   .@@@@@@@@@@@@@@@@@@@@---------=+++++++++|
|+---@@@@@@@@@@++++++++++@@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++======|
Variance from pattern = 1316374459
```
It is possible to get legible test pattern by reversing fix of issue #11 applied in https://github.com/igrr/esp32-cam-demo/commit/442ca8a2b7be4ffd765353a23222ccff995c2345

```
D (13968) camera: Waiting for positive edge on VSYNC
D (13978) camera: Got VSYNC
D (13978) camera: Waiting for frame
D (13988) camera: Frame done
D (13988) camera_demo: Done
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%#########@++++++++++==========:::::::::.         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%##########++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%**********++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%**********++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%**********++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%**********++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%**********++++++++++==========::::::::::         |
| @@@@@@@@@@@@@@@@@@@@%%%%%%%%%%**********++++++++++==========::::::::::         |
Variance from pattern = 69561395
```

The test pattern with XCLK at 10MHz with fix applied looks as follows::

```
D (5652) camera: Waiting for positive edge on VSYNC
D (5682) camera: Got VSYNC
D (5682) camera: Waiting for frame
D (5712) camera: Frame done
D (5712) camera_demo: Done
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%########## +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%********** +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%********** +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%********** +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%********** +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%********** +++++++++==========-:::::::::          |
|@@@@@@@@@@@@@@@@@@@@@%%%%%%%%%********** +++++++++==========-:::::::::          |
Variance from pattern = 6531449
```

Is it possible to have have #11  fixed and be able to operate OV7725 camera at 20MHz?

Application to test application in various scenarios including fix reversal is available under https://github.com/krzychb/esp32-cam-demo

igrr, 2016-12-30:
I have check with the hardware folks, an according to them, 20MHz is the highest signal frequency which can be captured by I2S when operating from an external clock. We'll try to debug this, but can't give any timeline — there are a few higher priority things which are keeping hardware team busy now.

igrr, 2017-01-23:
For the record, I did a bunch of tests yesterday and I can't get 20MHz to work. The reason why it "worked" previously was probably because the old code requested less data than what was being sent by the camera, making it somewhat tolerant to dat transfer errors. New code doesn't give slack on this, expecting all data to arrive (which makes sense in JPEG mode, for instance).

I'll look into this further but for now the highest supported frequency seems to be 16MHz.

igrr, 2017-01-24:
Status update: I now have a camera signal generator model inside a Lattice ICEStick, and it has helped a lot in narrowing down on the issue. At this point I'm fairly confident that at 20MHz one sample is lost at the end of HREF pulse due to a timing issue between pclk and href signals. Figuring out whether this is an issue of I2S peripheral or a side effect of the GPIO matrix is something we will do after the holidays.

In the mean time, I'll see if i can work around this, for example by setting "output PCLK when data valid" option in camera configuration registers, and connecting HREF to constant high value.

[image]

krzychb, 2017-01-24:
Cool idea to model the camera bus signals with FPGA.
Thanks for sharing the idea and update on progress!

Do you use some signals from real camera in parallel (judging from the photo) or all camera signals are entirely from the Lattice ICEStick?

igrr, 2017-01-25:
I kept the I2C connection to the camera to make the ESP32 code in `camera_init` happy. FPGA only generates the clock/href/vref and parallel data: https://gist.github.com/igrr/593e95297cc421b22f2aeec539ce8d6a

igrr, 2017-06-07:
20MHz now works fine on WROVER-KIT boards with OV7725. With OV2640 in JPEG mode there are occasional glitches visible, possibly due to signal integrity issues. I2S configuration issues related to 20MHz support have been fixed in #2.

closed 2017-06-07 by igrr