igrr/esp32-cam-demo #21: Do we need sdkconfig file in this repository?

krzychb, 2016-12-30:
As we keep development `sdkconfig` will soon contain:
```
CONFIG_WIFI_SSID=
CONFIG_WIFI_PASSWORD=
```
With ESP8266 development this was one of my problems to remember to remove these setting from code before posting on GitHub.

Now once we can detach user settings we should do it.

The only argument I see for keeping this file are potentially some application required settings like mentioned in https://github.com/igrr/esp32-cam-demo/issues/19.

Apart of strictly user specific ssid/pass, `sdkconfig` also contains `Serial flasher config` settings that are unlikely to be useful to other users. Anybody besides the author of last commit would need to change them anyway after cloning this repository.

Overall I believe we should remove it.

igrr, 2016-12-30:
We have an ESP-IDF feature coming up which can help here: support for `sdkconfig.defaults` file.
Similar to what we have in [05_ble_adv](https://github.com/espressif/esp-idf/tree/master/examples/05_ble_adv) example, just without any modifications to project Makefile. I think it will be merged soon, then we can use it here.

krzychb, 2016-12-30:
Thanks!
I was not aware of `sdkconfig.defaults`. This fits great here.

igrr, 2017-06-07:
sdkconfig has been removed and sdkconfig.defaults has been added. No more fighting about sdkconfig settings!

closed 2017-06-07 by igrr