igrr/esp32-cam-demo #24: How to make i2s isr occurs  after the acquisition of 10 rows of data instead of the acquisition of a line

jixunx, 2017-02-10:
It seems that i2s isr occurs  at the end of href.But in my project,I want i2s isr occurs after the acquisition of 10 rows of data.How to modify i2s configuration and camera configuration?I will appreciate it very much if you tell me!

igrr, 2017-02-10:
Actually the interrupt happens each time one DMA buffer is filled. So if you make a DMA buffer for 10 lines, you can have one interrupt per 10 lines. Note that the size of DMA buffer is limited to 12 bits (PITA, I know), so biggest you can have is 8188 bytes. If that fits 10 lines, you can use it.

In the latest code for jpeg branch, we actually can have multiple interrupts per line at higher resolutions.

jixunx, 2017-02-13:
In addition to buffer size multiply three times I use your code completely.But it seems that the i2s_isr did not take place after the full three lines, but still at the end of HREF.
I pull high GPIO15 and then pull down GPIO15 in i2s_isr .Analysing the timing of i2s_isr by logic analyzer.
[image]

I am sorry I am sorry I am unfamiliar with Git.Could you tell me where is  latest code for jpeg branch?

jixunx, 2017-02-14:
Hi igrr!
Can you answer my questions in spite of your busy schedule?
Thank you very much!

igrr, 2017-02-14:
Sorry, without seeing your modifications I can't say for sure what's wrong. In general, if you set RX_EOF_NUM and the DMA descriptor length to a certain value (making sure length doesn't exceed 4095), you will only get an interrupt once this given length of data is received.

jixunx, 2017-02-14:
Thank you!I did it!I have not set  RX_EOF_NUM!

closed 2017-06-07 by igrr