igrr/esp32-cam-demo #26: Send data collected from camera through wifi issue

jixunx, 2017-03-09:
Hi igrr!
   I use ov7670.OV7670 works pretty well.But i met issue about when I want to send out data collected through wifi.
   I ony modify camer_print_fb to sending data collected from camera:

```
void camera_print_fb()
{      
        gpio_set_level(33, 1);//gpio33 connects power down,pull high means turn off camera for save power.
	unsigned char start[5]="start";
	unsigned char end[3]="end";
	 int ret=0,ret_count=0;
	sendto(sock,start,SEND_START_LEN,0,(struct sockaddr*)&toAddr,sizeof(toAddr));
	for(int i=0;i<120;i++)
	{
				vTaskDelay(2/portTICK_RATE_MS);
				ret=sendto(sock,s_fb+i*480*2,960,0,(struct sockaddr*)&toAddr,sizeof(toAddr));
/*				
                                while(ret<0)
				{
				                vTaskDelay(2/portTICK_RATE_MS);
						gpio_set_level(17, 1);

						ret=sendto(sock,s_fb+i*480*2,960,0,(struct sockaddr*)&toAddr,sizeof(toAddr));
						gpio_set_level(17, 0);
						
				}
	}
*/
                         if(ret<0)
                         {
                                   ret_count++;
                         }
	ets_printf("ret_count=%d\n",ret_count);
	ret_count=0;
	sendto(sock,end,SEND_END_LEN,0,(struct sockaddr*)&toAddr,sizeof(toAddr));
	gpio_set_level(33, 0);//gpio33 connects power powen,pull low means turn on camera for data cololected.

}
```
    My network enviorment is good.Wifi LAN.There are only one sender and one recciever.Freertos's tick rate is 1000HZ.If I set delay time to 2 ms ,failure rate is high,failure reason if not enough spaces.So I need delay and retry many times. If I set delay time to 10ms,failure rate is low.
  I did the test that only make wifi transimit data and the has nothing to do with camera.2ms delay time is  enough,the failure rate is 0.
  In camera project,when data is being sent,camera power down,DMA off.I don't what reson causes the issue.
  By the way,I'd want to ask whether DMA's racing to control bus  can stop wifi core from getting data from memeory.Whether the electric current can infect wifi core.
  Thanks!
  Best regards!

jixunx, 2017-03-30:
Amoth ago.I disable PWM of esp32,and make the clock of camera from external crystal.The issue has fixed.Maybe PWM infect the wifi function.

krzychb, 2017-03-30:
Looks like a duplicate of #27

closed 2017-03-30 by krzychb