igrr/esp32-cam-demo #30: Display bitmap image over http

tuanpmt, 2017-06-03:
I've added httpd server and sent the captured bitmap file over http, some modified:
- RGB555 format: COM7_RES_VGA | COM7_FMT_RGB555 | COM7_FMT_RGB, FRAMESIZE_HQVGA
- line_filter_task: 
```cpp
uint32_t rgb = *buf;
rgb >>= 16; //remove 2Bytes i2s overhead
uint32_t red =       (rgb & 0b111110000000000) >> (10);
*pfb = red << 3;
pfb++;
// // // Set target pointer 8 bit forward
uint32_t green =     (rgb & 0b000001111100000) >> (5);
*pfb = green << 3;
pfb++;
uint32_t blue =      (rgb & 0b000000000011111);
*pfb = blue << 3;
pfb++;
buf ++;
```
- s_fb_size = s_fb_w * s_fb_h * `3`;

But it seems the data I received was not what I expected:

[image]

Could you please take a look at this:
https://github.com/tuanpmt/esp32-cam-demo

Thanks.,

igrr, 2017-06-03:
iirc that is not the layout of the bytes produced by i2s in camera mode.

I made a few notes in the JPEG branch:

https://github.com/igrr/esp32-cam-demo/blob/43343bbf7057c1bc217a024d2aba91c2b60e9985/components/camera/camera_common.h#L25-L38

https://github.com/igrr/esp32-cam-demo/blob/43343bbf7057c1bc217a024d2aba91c2b60e9985/components/camera/camera.c#L537

As you can see, the image is skewed to the right, which usually happens when the stride which you use to read the input buffer is less than the real stride of the buffer. Also the skew is not constant, it is less for some lines, which may indicate that you are loosing some samples (might be a timing issue). Adding some asserts in the code to check whether you are missing some buffers may help.

tuanpmt, 2017-06-04:
I have switched to the i2s mode as you say, but still not as desirable. Although the picture is different 😄 . I will refer to the `feature/ov2460` branch and the `openmv` repository
Thank @igrr

igrr, 2017-06-07:
I'll keep this open as displaying RGB image in the browser looks like a nice feature for the demo.

tuanpmt, 2017-06-08:
I've updated the repository, and add bitmap filter: https://github.com/igrr/esp32-cam-demo/compare/master...tuanpmt:master

But still not work:
[image]

igrr, 2017-06-08:
Okay, after some fixes this works for me on a WROVER-KIT board with OV7725:
https://github.com/igrr/esp32-cam-demo/tree/feature/rgb_bitmap

(the image is flipped, but that should be easy to fix)
[image]

tuanpmt, 2017-06-10:
I've tested branch `feature/rgb_bitmap` with enable test pattern 
[image]

and no:
[image]

Are there any diagnoses?

igrr, 2017-06-12:
That's quite similar to the picture I got above, I had assumed that such posterized image is normal for RGB565 mode without any dithering. Maybe there still is some bug in color conversion code.
I think we should set up a test to check that there are indeed 32 levels of red and blue resolved, and 64 levels of green. Unfortunately I don't have any color filters at hand, but i'll see what i can get on taobao.

igrr, 2017-11-08:
Merged in #33

closed 2017-11-08 by igrr