igrr/esp32-cam-demo #37: Ov2640 reset eroor on soft reboots

donny681, 2017-06-14:
Why ov2640 reset eroor on soft reboots.I use logic analyzer to catch signal,it seems that the camera don't respose ack after soft reboots.And how to fix it.

>Getting an error during i2c check of camera:
D (1843) camera: Detected camera at address=0x30
D (1853) camera: Camera PID=0x00 VER=0x00 MIDL=0x00 MIDH=0x00

igrr, 2017-06-14:
I think this was previously mentioned in another issue:
https://github.com/igrr/esp32-cam-demo/issues/15#issuecomment-278886227

For some reason I haven't had issues with reset though. Maybe @Oitzu has some idea.

donny681, 2017-06-21:
@igrr I add GPIOs interrupt to the project and find that GPIO can't go to the isr after getting a frame.Before getting a frame ,GPIOs  can go to the isr.I try to register isr again  but it is no use.

donny681, 2017-06-27:
@igrr @Oitzu  I fix the bug that the ov2640 reset error.The reg(0xff) must be programmed 0x01 after reseting by pulling down the rst.
`gpio_set_level(config->pin_reset, 0);
    delay(10);
    gpio_set_level(config->pin_reset, 1);
    delay(10);

    uint8_t buf[] = {0xff, 0x01};
            twi_writeTo(0x30, buf, 2, true);

            delay(100);

    ESP_LOGD(TAG, "Searching for camera address");
    /* Probe the sensor */
    delay(10);
    uint8_t slv_addr = SCCB_Probe();
    if (slv_addr == 0) {
        *out_camera_model = CAMERA_NONE;
        return ESP_ERR_CAMERA_NOT_DETECTED;
    }`

donny681, 2017-06-27:
@igrr @Oitzu I fix the bug that the ov2640 reset error.The reg(0xff) must be programmed 0x01 after reseting by pulling down the rst.
`gpio_set_level(config->pin_reset, 0);
    delay(10);
    gpio_set_level(config->pin_reset, 1);
    delay(10);
    /*ater reseting the senor,the reg (0xff) must be programmed 0x01*/
    uint8_t buf[] = {0xff, 0x01};
            twi_writeTo(0x30, buf, 2, true);
            delay(100);
    ESP_LOGD(TAG, "Searching for camera address");
    /* Probe the sensor */
    delay(10);
    uint8_t slv_addr = SCCB_Probe();
    if (slv_addr == 0) {
        *out_camera_model = CAMERA_NONE;
        return ESP_ERR_CAMERA_NOT_DETECTED;
    }`

donny681, 2017-06-27:
@igrr @Oitzu I fix the bug that the ov2640 reset error.The reg(0xff) must be programmed 0x01 after reseting by pulling down the rst.
`gpio_set_level(config->pin_reset, 0);
    delay(10);
    gpio_set_level(config->pin_reset, 1);
    delay(10);
    /*ater reseting the senor,the reg (0xff) must be programmed 0x01*/
    uint8_t buf[] = {0xff, 0x01};
            twi_writeTo(0x30, buf, 2, true);
            delay(100);
    ESP_LOGD(TAG, "Searching for camera address");
    /* Probe the sensor */
    delay(10);
    uint8_t slv_addr = SCCB_Probe();
    if (slv_addr == 0) {
        *out_camera_model = CAMERA_NONE;
        return ESP_ERR_CAMERA_NOT_DETECTED;
    }
    s_state->sensor.slv_addr = slv_addr;
    ESP_LOGD(TAG, "Detected camera at address=0x%02x", slv_addr);`

donny681, 2017-06-27:
@igrr @Oitzu I fix the bug that ov2640 reset errror. The  reg(0xff) of the sensor must be programmed 0x01 after reset the camera.
`   gpio_set_level(config->pin_reset, 0);

    delay(10);

    gpio_set_level(config->pin_reset, 1);

    delay(10);

    /*after reseting the sensor,the reg (0xff) must be programmed 0x01*/
    uint8_t buf[] = {0xff, 0x01};
            twi_writeTo(0x30, buf, 2, true);
            delay(100);
    ESP_LOGD(TAG, "Searching for camera address");
    /* Probe the sensor */
    delay(10);
    uint8_t slv_addr = SCCB_Probe();
    if (slv_addr == 0) {
        *out_camera_model = CAMERA_NONE;
        return ESP_ERR_CAMERA_NOT_DETECTED;
    }
    s_state->sensor.slv_addr = slv_addr;
    ESP_LOGD(TAG, "Detected camera at address=0x%02x", slv_addr);`

lss6378, 2017-11-03:
@donny681 
Think you so much! It works!

sravanth005, 2019-04-04:
in the ESP32 Ai thinker CAM there is a reset button does any one know what signal does reset button generates