igrr/esp32-cam-demo #38: Read frame data

hechao, 2017-06-16:
Hello!

Very cool project! Thanks for making this.

I have tried the demo, but the code seems updated, the question is how to read the data downloaded from http://192.168.0.104/get

I have tried to rename the file from http://192.168.0.104/pgm to pgm.pgm, and open it with a tool called openseeit, it is very grey rough image. This is a problem of the camera or what could be?
[image]

igrr, 2017-06-16:
If you are using OV7725 in grayscale mode, `http://your-ip/get` returns raw image (without any headers), one byte per pixel. `/pgm` should return the same, plus PGM header. I can't really tell whether this result you get is due to an issue with hardware, code, or the viewer.
Can you attach this PGM file you get, or post a link to it?

hechao, 2017-06-16:
Yes, sure, thank you very much for the swift reply.

Using 7725 camera, hardware data pin is D2-D7 (D0 D1 should be not used right?), Reset pin set to IO15, the rest is fully same.
[pgm.zip](https://github.com/igrr/esp32-cam-demo/files/1080428/pgm.zip)

Also when I turn on "[ ] Enable test pattern on camera", and also OV7725 and OV2640, the test pattern not working. One more question, is OV2640 also supported now?

Thank you!!

igrr, 2017-06-16:
The image looked posterized, so i checked the histogram. Here is the histogram of image bytes:
[image]

See that periodic structure? There are ranges of values, each is 16 units wide. This doesn't look like a random error.

Here is a histogram of *bits* in your image. It makes me think that two wires/pins from the camera are not connected. If i'm not mistaken, that's D7 and D6:

[image]

igrr, 2017-06-16:
OV2640 is supported now, although it is still has some random glitches in JPEG mode at 20MHz (might be signal integrity issue on WROVER-KIT board).

W.r.t. test pattern not working, can you give more info? Any error log? If there is no error, do you get some image instead?

hechao, 2017-06-16:
Thanks, will check this and let you know later.

hechao, 2017-06-19:
need some time to check it, will close it now, thanks for support

closed 2017-06-19 by hechao