igrr/esp32-cam-demo #39: When esp32 sets up modem sleep, i2s's isr is in a mess。

jixunx, 2017-06-23:
The IDF version I use is the rc2.0 version. Network environment is WiFi lan.
I use esp32+camera to do network real-time video transmission. When esp32 grab a frame of data from camera, it sends out a frame of data through the wifi. When the esp32's WiFi finished transimission of one frame , it goes into modem sleep. If I set modem sleep mode, the i2s's isr  becomes abnormal after the system has been running for some time. But if I don't set modem sleep mode, the system will work normal all the time.
Here is a log of logic analyser
[image]

igrr, 2017-06-23:
Thanks for reporting this.
- Does this happen with the current master branch of this project? 
- Does this happen with latest ESP-IDF?
- What type of camera and what mode/configuration do I need to set to reproduce this?

jixunx, 2017-06-23:
Does this happen with the current master branch of this project?
It's not the currnet master branch.Esp32-cam-demo's master version is before 2017.
Does this happen with latest ESP-IDF?
I have not test it with latest ESP-IDF.
What type of camera and what mode/configuration do I need to set to reproduce this?

[sdkconfig.zip](https://github.com/igrr/esp32-cam-demo/files/1096737/sdkconfig.zip)

igrr, 2017-06-23:
Ok, would you mind testing with more recent versions of both esp32-cam-demo and esp-idf? I've been running the latest version of both, with modem sleep enabled, on OV2640 for the last 15 minutes, and i'm not seeing any problem. Chances are, the bug you are referring to has been fixed at some point.

jixunx, 2017-06-23:
I suggest you test it for thirty minutes at a time and test more times．The bug is not happen 100%.

jixunx, 2017-06-23:
Testing with recent esp-idf is easy,but test with recent esp32-cam-demo is not easy and maybe unnecessary.The version of esp32-cam-demo I use is" commit 442ca8a2b7be4ffd765353a23222ccff995c2345".It is commited on  13 Dec 2016.It seems that the updates after the commit have nothing to do with my project except "camera: fix typo in LC_CONF bit masks".It's about ov2640 and http not about i2s.I don't need these updates.That is to say,recent version of esp-cam-demo does not apply to my project.

jixunx, 2017-06-23:
Ｉhave test it with esp-if v2.0 for two times.The bug happens once.After running for forty minutes, The bug happens.