igrr/esp32-cam-demo #41: OV2640 can not work right!

yjl, 2017-07-21:
@igrr Here is  verbose log!  it is seem like work well ,but image is not right in the browser.
  And could you update Readme about the Ov2640,thank you.
`
I (1222) cpu_start: Pro cpu up.
I (1234) cpu_start: Starting app cpu, entry point is 0x40080eb0
I (0) cpu_start: App cpu up.
I (1267) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1289) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1310) heap_alloc_caps: At 3FFB79E8 len 00028618 (161 KiB): DRAM
I (1331) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1352) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1374) heap_alloc_caps: At 40093C7C len 0000C384 (48 KiB): IRAM
I (1395) cpu_start: Pro cpu start user code
I (1454) cpu_start: Starting scheduler on PRO CPU.
I (206) cpu_start: Starting scheduler on APP CPU.
D (206) nvs: nvs_flash_init_custom start=9 count=6
D (246) camera: Enabling XCLK output
D (246) ledc: LEDC_PWM CHANNEL 0|GPIO 21|Duty 0001|Time 0
D (246) camera: Initializing SSCB
D (246) camera: Resetting camera
D (266) camera: Searching for camera address
D (276) camera: Detected camera at address=0x30
D (286) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
D (286) camera: Doing SW reset of sensor
I (326) camera_demo: Detected OV2640 camera, using JPEG format
D (366) camera: Setting frame size to 640x480
D (426) camera: Test pattern enabled
D (426) camera: in_bpp: 2, fb_bpp: 2, fb_size: 61440, mode: 0, width: 640 height: 480
D (426) camera: Allocating frame buffer (61440 bytes)
D (426) camera: Initializing I2S and DMA
D (436) camera: Line width (for DMA): 5120 bytes
D (436) camera: DMA buffer size: 2560, DMA buffers per line: 2
D (446) camera: DMA buffer count: 8
D (446) camera: Allocating DMA buffer #0, size=2560
D (446) camera: Allocating DMA buffer #1, size=2560
D (456) camera: Allocating DMA buffer #2, size=2560
D (456) camera: Allocating DMA buffer #3, size=2560
D (466) camera: Allocating DMA buffer #4, size=2560
D (466) camera: Allocating DMA buffer #5, size=2560
D (476) camera: Allocating DMA buffer #6, size=2560
D (476) camera: Allocating DMA buffer #7, size=2560
D (486) camera: Initializing GPIO interrupts
D (576) camera: Init done
D (576) nvs: nvs_open misc 1
D (576) nvs: nvs_get_str_or_blob log
I (576) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (576) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (586) nvs: nvs_open nvs.net80211 1
D (596) nvs: nvs_get opmode 1
D (596) nvs: nvs_get country 1
D (596) nvs: nvs_get_str_or_blob sta.ssid
D (606) nvs: nvs_get_str_or_blob sta.mac
D (606) nvs: nvs_get sta.authmode 1
D (616) nvs: nvs_get_str_or_blob sta.pswd
D (616) nvs: nvs_get_str_or_blob sta.pmk
D (616) nvs: nvs_get sta.chan 1
D (626) nvs: nvs_get auto.conn 1
D (626) nvs: nvs_get bssid.set 1
D (626) nvs: nvs_get_str_or_blob sta.bssid
D (636) nvs: nvs_get sta.phym 1
D (636) nvs: nvs_get sta.phybw 1
D (636) nvs: nvs_get_str_or_blob sta.apsw
D (646) nvs: nvs_get_str_or_blob sta.apinfo
D (646) nvs: nvs_get_str_or_blob ap.ssid
D (656) nvs: nvs_get_str_or_blob ap.mac
D (656) nvs: nvs_get_str_or_blob ap.passwd
D (666) nvs: nvs_get_str_or_blob ap.pmk
D (666) nvs: nvs_get ap.chan 1
D (666) nvs: nvs_get ap.authmode 1
D (676) nvs: nvs_get ap.hidden 1
D (676) nvs: nvs_get ap.max.conn 1
D (676) nvs: nvs_get bcn.interval 2
D (686) nvs: nvs_get ap.phym 1
D (686) nvs: nvs_get ap.phybw 1
D (686) nvs: nvs_get ap.sndchan 1
D (696) nvs: nvs_set_blob sta.mac 6
D (696) nvs: nvs_set_blob ap.mac 6
D (706) nvs: nvs_open phy 0
D (706) nvs: nvs_get cal_version 4
D (706) nvs: nvs_get_str_or_blob cal_mac
D (706) nvs: nvs_get_str_or_blob cal_data
D (716) nvs: nvs_close 3
I (736) phy: phy_version: 350, Mar 22 2017, 15:02:06, 0, 0
I (736) camera_demo: Connecting to "CHQJ_DEV"
D (736) tcpip_adapter: check: local, if=0 fn=0x4010d644

D (736) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (746) tcpip_adapter: check: remote, if=0 fn=0x4010d644

D (3096) tcpip_adapter: check: local, if=0 fn=0x4010d820

D (3096) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (3096) tcpip_adapter: check: remote, if=0 fn=0x4010d820

D (3106) tcpip_adapter: check: local, if=0 fn=0x4010db9c

D (3116) tcpip_adapter: dhcp client init ip/mask/gw to all-0
D (3116) tcpip_adapter: dhcp client start successfully
D (3126) tcpip_adapter: call api in lwip: ret=0x0, give sem
D (3126) tcpip_adapter: check: remote, if=0 fn=0x4010db9c

I (4116) event: ip: 192.168.0.100, mask: 255.255.255.0, gw: 192.168.0.1
I (4116) camera_demo: Connected
I (4116) camera_demo: Free heap: 153896
I (4116) camera_demo: Camera demo ready
I (4116) camera_demo: open http://192.168.0.100/get for single frame
I (4126) camera_demo: open http://192.168.0.100/stream for multipart/x-mixed-replace stream (use with JPEGs)
D (10766) camera_demo: Image requested.
D (10766) camera: Waiting for positive edge on VSYNC
D (10806) camera: Got VSYNC
V (10806) camera: gpio isr, cnt=0
D (10806) camera: Waiting for frame
V (10816) camera: isr, cnt=1
V (10816) camera: isr, cnt=2
V (10826) camera: isr, cnt=3
V (10836) camera: isr, cnt=4
V (10836) camera: isr, cnt=5
V (10846) camera: isr, cnt=6
V (10856) camera: isr, cnt=7
V (10856) camera: isr, cnt=8
V (10866) camera: isr, cnt=9
V (10866) camera: isr, cnt=10
V (10876) camera: isr, cnt=11
V (10886) camera: isr, cnt=12
V (10886) camera: isr, cnt=13
V (10896) camera: isr, cnt=14
V (10896) camera: gpio isr, cnt=14
I (10896) camera: Frame 0 done in 135 ms
D (10896) camera_demo: Done
D (13956) camera_demo: Image requested.`
[image]

igrr, 2017-11-07:
I suggest trying with bitmap mode first. See if you get reasonable and stable image. The issue in JPEG mode might be caused by data corruption. It is easier to debug the setup and connections in bitmap mode first, and then go to JPEG mode.

James-sjec, 2017-11-16:
where to enable bitmap mode