igrr/esp32-cam-demo #45: How to set OV7725 to get RGB565 picture

z43740979, 2017-07-27:
I set the OV7725 register for the RGB565 mode but the BMP picture data is still gray data, please explain the source of s_fb data 
I’d appreciate your help

igrr, 2017-11-07:
The data returned by the camera is passed into the line filter, as explained in this section:
https://github.com/igrr/esp32-cam-demo#operation

RGB565 support is being added in the PR: https://github.com/igrr/esp32-cam-demo/pull/33. Perhaps you can check out that branch and give a try.

igrr, 2017-11-08:
You can now use master branch, RGB565 mode is used by default for OV7725. Can be switched to YUV here: https://github.com/igrr/esp32-cam-demo/blob/master/main/app_main.c#L65.

closed 2017-11-08 by igrr

z43740979, 2017-11-08:
OK I got the RGB picture with your code ~~~Its really helpful