igrr/esp32-cam-demo #46: Why appeared reset problem after Multiply the camera clock ？

FlyingArm, 2017-08-14:
My project's idf is esp-idf-v2.0.
But it has some problems now.
If camera pclk is 10MHZ,and set i2s clkm_div_num = 2 ,it can normal work.
This is the set i2s mclk code,the camera's pclk was seen with a logic analyzer and is 10MHZ.
    I2S0.clkm_conf.clkm_div_a = 1;
    I2S0.clkm_conf.clkm_div_b = 0;
    I2S0.clkm_conf.clkm_div_num = 2;
But if set camera pclk is 20MHZ,and set i2s clkm_div_num = 1,there are aproblem .It will print the following and reset .
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400fcb8d  PS      : 0x00060f30  A0      : 0x800f1d74  A1      : 0x3ffd8390  
A2      : 0x00000000  A3      : 0x3ffdfd88  A4      : 0x00000058  A5      : 0x3ffdfdb8  
A6      : 0x3ff42000  A7      : 0x00000000  A8      : 0x800fcb6d  A9      : 0x3ffd8376  
A10     : 0x3ffdfdf0  A11     : 0x00000058  A12     : 0x00000058  A13     : 0x00000058  
A14     : 0xdbefe9e9  A15     : 0x999f21a2  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x999f21be  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400fcb8d:0x3ffd8390 0x400f1d71:0x3ffd83c0 0x400f1ae2:0x3ffd83e0 0x400eb72a:0x3ffd8410 0x400ed3c0:0x3ffd8470 0x400efbcf:0x3ffd84b0 0x400efc4d:0x3ffd84d0 0x400edd14:0x3ffd84f0 0x400edd33:0x3ffd8530 0x400ef883:0x3ffd8550 0x400ef966:0x3ffd8570 0x400f86fb:0x3ffd8590
This is the set i2s mclk code,the camera's pclk was seen with a logic analyzer and is 20MHZ.
I2S0.clkm_conf.clkm_div_a = 1;
I2S0.clkm_conf.clkm_div_b = 0;
I2S0.clkm_conf.clkm_div_num = 1;

I don't why doubling the rate of both will cause problems ?
If i want to improve the camera's acquisition rate,How do I set up I2S ?

igrr, 2017-08-14:
I have just checked and setting div_num=1 will not work — div_num=2 is the lowest setting which the hardware supports.

Sampling at 20MHz can only be achieved (as this is done in this project) by setting I2S0.fifo_conf.rx_fifo_mod to 0 (defined as `SM_0A0B_0B0C` in the source code). Please read this comment: https://github.com/igrr/esp32-cam-demo/issues/29#issuecomment-307871635, especially the paragraph starting from "Why do we need this odd SM_0A0B_0B0C mode though?".

With regards to why the exception happens, i can not say — the backtrace contains useful information which you should be able to interpret if you use `make monitor` or if you enable GDB stub. Either way you can get a readable backtrace. With GDB you can also figure out which variable/memory location was corrupted. Good luck.

closed 2017-08-14 by igrr

FlyingArm, 2017-08-14:
Thank you for your answer.