igrr/esp32-cam-demo #48: I can't see anything in http browser

JPMJPM, 2017-09-14:
Hi,  the log seem to be ok but I can't see anything in the web browser (http://192.168.1.34/stream).

Any clue ?, Regards. 

I (312) camera_demo: Detected OV2640 camera, using JPEG format
I (532) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (532) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (572) phy: phy_version: 359.0, e79c19d, Aug 31 2017, 17:06:07, 0, 0
I (582) camera_demo: Connecting to "emc2"
I (3022) event: ip: 192.168.1.34, mask: 255.255.255.0, gw: 192.168.1.1
I (3022) camera_demo: Connected
I (3022) camera_demo: Free heap: 152548
I (3022) camera_demo: Camera demo ready
I (3032) camera_demo: open http://192.168.1.34/get for single frame
I (3032) camera_demo: open http://192.168.1.34/stream for multipart/x-mixed-replace stream (use with JPEGs)
I (7082) camera: Frame 0 done in 152 ms
I (18752) camera: Frame 1 done in 153 ms
I (40262) camera: Frame 2 done in 152 ms
I (61292) camera: Frame 3 done in 157 ms
I (103442) camera: Frame 4 done in 83 ms
I (115512) camera: Frame 5 done in 150 ms
I (118712) camera: Frame 6 done in 148 ms
I (139102) camera: Frame 7 done in 149 ms
I (159812) camera: Frame 8 done in 155 ms
I (168132) camera: Frame 9 done in 134 ms
I (174772) camera: Frame 10 done in 103 ms
I (186842) camera: Frame 11 done in 139 ms

igrr, 2017-11-07:
Perhaps there will be some useful info in the browser developer panel? In Chrome, you can have a look at network tab there and see what the browser has to say about the requests being completed or not.

One possibility is that the JPEG data is so horribly broken that the browser can't even start decoding the image. Could you try grabbing bitmaps with the same wiring and see if you get stable image or not?

JPMJPM, 2017-11-07:
Thanks, now everything is ok. My project progresses little by little in a satisfactory way.

closed 2017-11-08 by igrr