igrr/esp32-cam-demo #50: Failed to allocate frame buffer

black5heep, 2017-09-28:
Can get the maximum resolution of 220x160 and then a little more memory application failed
How to get resolution of 640x480

igrr, 2017-11-07:
There was a regression in ESP-IDF some time ago that caused extra 20kB to be allocated at startup. Could you please update the IDF and try again?

closed 2017-11-17 by igrr

thing91, 2018-02-05:
Which version of esp-idf eliminates this problem ? I use 3.0 and 2.1 and problem is still appearing.

eg-astrouka, 2018-02-13:
@thing91 try release v2.1 (not 2.1.1)

thing91, 2018-02-14:
I did try but when I run "make flash" i get bunch of errors connected to **cxx** component and gthr-default.h in xtensa toolchain.

eg-astrouka, 2018-02-14:
@thing91 I haven't tried this code, but I'm using https://github.com/tekker/esp32-ov7670-hacking without any problems