igrr/esp32-cam-demo #51: Cannot capture images using ov7725 in esp32

igrr, 2017-11-07:
_From @James-sjec on November 7, 2017 14:43_

example code loaded : https://github.com/igrr/esp32-cam-demo

Following error occured when started : 
![apc - 2017 11 07 20 05 - 001 3d](https://user-im

ages.githubusercontent.com/32930819/32498994-1a70dbde-c386-11e7-995e-552dfd0f4fa3.jpg)

 

_Copied from original issue: espressif/esp-idf#1240_

igrr, 2017-11-07:
Are you seeing this issue in the original code, or have you modified anything?

Also please attach your sdkconfig file.

James-sjec, 2017-11-07:
"GPIO_PIN mask error" was solved , because I made changes in the pins of "ESP32 Camera  Demo Configuration".

Still I am not able to capture the images.
I typed 192.168.43.150/pgm in browser url , I got pgm file and  add .pgm extension and opened it , I got the following image
[image]
 
I opened the image using Inkspace software 
Is there any photo viewer to view those images or Is the camera  not capturing the images

I also tried 192.168.43.150/stream I do not see anything on the browser expect a black screen
[image]

Following is the sdkconfig file

[sdkconfig.txt](https://github.com/igrr/esp32-cam-demo/files/1451346/sdkconfig.txt)

igrr, 2017-11-08:
Indeed that looks seriously broken! Are you using the camera header on WROVER-KIT or are you wiring the camera to another development board? Have you tried enabling the test pattern on the camera? You can do that using an option in menuconfig. It might be easier to figure out what is going on here with the test pattern enabled.

James-sjec, 2017-11-08:
I am using WROOM-32 , below is the picture of the development board
[image]

I enabled test pattern ,after which I got the following error
[image]
[image]

I am using following pin map
[image]

AlanParsonsSHP, 2018-10-30:
Hi James,

I'm trying to use the ov7725 as well, but I get some errors at compilation time.
Do you wish to share your code? in order to compare.
Any help would be appreciated.
Many thanks in advance.

Alan