igrr/esp32-cam-demo #52: Limitations of pin-mappings?

Oitzu, 2017-11-11:
I was looking into etching some board for my testrig and was wondering are there some limitation about the pin-mappings i can do?
Documentation just states "Pinout can be adjusted to some extent in software.", but not the limits of these.

igrr, 2017-11-11:
This remark is about the case when cameras with different pinouts are plugged into WROVER-KIT camera header. Because some of the pins routed to this header are input-only (34-39), this means that not all pinouts can be accommodated.

If you are making your own board, ESP32 datasheet is your friend — consider that some pins are input only, some are used for bootstrapping, some are used for JTAG and so on. Feel free to post the pinout here and I'll sanity-check it.

Oitzu, 2017-11-11:
Thank you, very helpful as ever. :)

> If you are making your own board,
Nope not a full-fledged board just a single sided board to connect my ESP-WROOM-32 and the camera safely together, to ease testing.

I was thinking about following pinout to avoid jumper wires, on the single sided board, on the signal wires.:

```
PWDN - x
D2 - IO22
D4 - IO21
D6 - IO19
D8 - IO18
CXLK - IO5
HREF - IO17
SIOD - IO16

RESET (Any free I/O-Pin)
D3 - SVP
D5 - SVN
D7 - IO34
D9 - IO35
PCLK - IO25
VSYNC - IO26
SIOC - IO27
SIOC - IO27
```

igrr, 2017-11-13:
Don't see any potential issues. I think this pinout should work fine.

igrr, 2017-11-13:
That actually gives me an idea to put a PCB equivalent of https://github.com/igrr/esp32-cam-demo/blob/master/pictures/ov7725-alternate-wiring.png onto dirtypcbs.com or OSHPark, so that folks who need can order the same design.

Oitzu, 2017-11-13:
> Don't see any potential issues. I think this pinout should work fine.

Ok thanks. :+1: 
Would be following, in shot, correct?

> SIOC, SIOD needs (to be): I/O
> Camera Reset, XCLK needs: O
> VSYNC, HREF, PCLK, D2-9 needs: I

> That actually gives me an idea to put a PCB equivalent...

Good idea.

igrr, 2017-11-14:
That's correct, but also mind the bootstrapping pins of ESP32 (0, 2, 12).

igrr, 2017-11-23:
@Oitzu Done! Here's the breakout: https://dirtypcbs.com/store/designer/details/igrr/5842/cam-breakout-v1-zip.

Note that i forgot to add footprints for pullup resistors on SDA and SCL, to my defence it seems that the modules already come with some internal pullups, so overall this works fine.

Oitzu, 2017-11-23:
Fabulous. :+1:

Oitzu, 2018-01-28:
What do you think about 
```
(13) D0      
(14) D1      
(12) D2      
(26) D3      
(27) D4      
(33) D5      
(25) D6      
(35) D7      
(32) XCLK    
(39) PCLK    
(34) VSYNC   
(36) HREF    
(21) SDA     
(22) SCL     
(15) RESET   
```
? (Taken out of menuconfig)

I'm atm trying to get the code running on the esp32-wrover but getting multiple resets after
`
D (161) camera: Detected camera at address=0x30
D (171) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
D (171) camera: Doing SW reset of sensor
ets Jun  8 2016 00:22:57

rstets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
`

Oitzu, 2018-01-28:
(13) D0  
Seems to be the offending one.. but no idea why, even after reading the datasheet multiple times.

igrr, 2018-01-28:
My initial thought was about GPIO12, which is the flash voltage bootstrapping pin. GPIO13 is not a bootstrapping pin and should be unused by PSRAM. Really odd that it causes a rest. By the way, what is the exception/reset message printed?

Oitzu, 2018-01-28:
Checked again. The Problem stopped as soon as i cuted the line to gpio13. With gpio13 connected the Board resetted as soon as the ov2640 was soft resetted (and probably starting to bang bits out of D2).

Oitzu, 2018-01-28:
But maybe there is an miswiring/wrong label on the breakout Board. Unlikely but ist will check that tommorow evening.

Oitzu, 2018-01-29:
Problem is only present at init. If IO13 is connected after init, the esp32 does not reset.
The i2s interrupt doesn't fire but this is probably some other Problem.