igrr/esp32-cam-demo #54: Problem with esp_log in camera.c

x3mx3mal, 2017-11-16:
There is no ESP_LOG  messages from camera.c in monitor, only from app_main. How to fix this?

igrr, 2017-11-17:
Could you please give a bit more details?
I can't reproduce this issue with the latest master. Here's what i get:

```
I (5959) camera_demo: Free heap: 194740
I (5959) camera_demo: Camera demo ready
I (15359) camera: Frame 0 done in 105 ms
I (32069) camera: Frame 1 done in 156 ms
I (33349) camera: Frame 2 done in 104 ms
I (34389) camera: Frame 3 done in 139 ms
I (39029) camera: Frame 4 done in 89 ms
I (40389) camera: Frame 5 done in 155 ms
I (41589) camera: Frame 6 done in 137 ms
I (44229) camera: Frame 7 done in 103 ms
I (45349) camera: Frame 8 done in 137 ms
I (46469) camera: Frame 9 done in 86 ms
```

Logs from camera.s can be seen here.

igrr, 2017-12-21:
Closing due to no feedback.

closed 2017-12-21 by igrr