igrr/esp32-cam-demo #55: Unable to see the test pattern on the serial monitor

James-sjec, 2017-11-22:
I get the following error on screen : 
[image]

James-sjec, 2017-11-22:
And how to Open and View the get , pgm and   x-portable-graymap files

igrr, 2017-11-23:
Hmm, it seems that the documentation has fallen out of date with respect to the code.
The demo no longer prints ASCII art output on the console, instead it starts up a web server, and you can see images in web browser (URLs are printed at startup).

I suggest setting log level to the default one (INFO), you seem to have set it to VERBOSE. It seems that when log from ISR is enabled, the application can not keep up with the timing of the frame, which eventually causes lines to be dropped.

James-sjec, 2017-11-24:
Thank you Sir!
I changed log level to INFO mode , and the above issue was solved.

I (5339) camera_demo: Camera demo ready
I (5349) camera_demo: open http://192.168.1.30/get for single frame
I (5349) camera_demo: open http://192.168.1.30/pgm for a single image/x-portable-graymap image

I typed in the browser as "http://192.168.1.30/pgm" . A  file named 'pgm' got downloaded I added .pgm extension and the image I got the following image : 

[image]

I typed in the browser as "http://192.168.1.30/get" . A file named 'get' got downloaded I added .pgm  extension and tried opening it ,  Image viewers says unsupported format . The same goes to http://192.168.1.30//x-portable-graymap 

how to open and view these files , and please help me in capturing the images.

I use ESP-WROOM-32 DEVKITV1 development board and OV7725 camera.
And tried enabling test pattern i am getting the same result.

igrr, 2017-11-24:
Based on the image, it looks like there is some serious noise on the lines
between the ESP and the camera. How long are the wires connecting the
camera to the ESP?

You can try reducing clock frequency (it is 20MHz by default), see camera
demo configuration in menuconfig. Try something lower, e.g. 4MHz, and see
if the pgm image becomes more clear.

jfbrazeau, 2018-01-27:
Hello,

Thank you for this great sample "esp32-cam-demo". I have bought a OV7725 camera and I tried to use it with an ESP32 using this demo.

By default in PGM mode, I get something like this :
[image]

If I set `CONFIG_ENABLE_TEST_PATTERN` parameter to `y`, I get something like this :
[image]

To answer one of your question below, I don't think that the wires are too long as you can see (but if you think it's too long let me know !) :
[image]
Note : I've chosen the `Alternate ESP32 Pin Mapping` that you describe [here](https://github.com/igrr/esp32-cam-demo#connect)  (and I have checked the gpio pins connection several times... so that I am almost sure that everything is rightly connected, except the 2.2uF capacitor connected between GND and EN as I understood that the problem is for windows users, and I use a mac ; let me know if I'm wrong).

I tried to set the clock frequency to 4 Mhz, as you mentioned, but it leads to the following error (`SCCB` interface communication problem) :
[image]

I saw [here](https://cdn.sparkfun.com/datasheets/Sensors/LightImaging/OV7725.pdf) that the min clock value is 10 Mhz ? So I used 10 Mhz, and the result is intersting becauseit seems that the result I get is linked to the clock value (still with `CONFIG_ENABLE_TEST_PATTERN` enabled) ; with 10 Mhz :
[image]

The result changes if I increase the clock frequency, but for a given frequency, I almost get the same result each time I call the HTTP URL. For example, with 14 Mhz :
[image]

I saw that you talked about "serious noise" on the lines. Do you think it is really a problem of noise ? I thought that with a problem of noise, the "vertical shaded bars" of the "test pattern mode" should be at the right position and erroneous colors would appear randomly. Here it looks like the test pattern is there but shifted in a manner that is linked to the clock frequency.

Thank you in advance if you find a little time to give me your opinion about this issue...

Note : same issue in BMP mode (RGB565 / QQVGA) :
[image]

krzychb, 2018-01-27:
Hello @jfbrazeau,

The wires on the picture above are couple of times longer than recommended for the OV7725 and ESP32 DevKitC. Below is what worked for me:

[image]

Please read through the [Showcase](https://github.com/igrr/esp32-cam-demo#showcase) section that describes real tests made with OV7725.

jfbrazeau, 2018-01-28:
Ok thank you very much for the feedback. I have to buy connectors and wires to test like this. This will take a few days (or weeks). I'll tell you if it solves this issue. Thank you.

James-sjec, 2018-02-03:
Hai ,
 I soldered camera and esp32 to bread board , and the above issue was solved 
 I getting the following images  : 

[image]

How to increase the quality of the image

jfbrazeau, 2018-02-24:
Hello,

I come back here to give you a feedback. Approximately two weeks ago, I tried again with shorter wires as you told me. The result was worst than with my initial wires. I think that my wires were very bad.

Today I soldered the circuit, and... everything works fine now (as you can see below) !

Thank you very much for your help ! (and for this great sample project on github)

Regards,

Jean-François Brazeau

[image]

[image]