igrr/esp32-cam-demo #56: Compile Error

LilyGO, 2017-12-11:
Hi, I have a hard time. I do not know why compilation failed here because it's a well-known code. Maybe I have less configuration. I try to comment out this line of code, compiled successfully. Can be uploaded minicom display simply can not start the camera. I hope everyone can help me solve this problem. Thank you.

code:

/home/lee/work/12.11/esp32-cam-demo/components/camera/./xclk.c: In function 'camera_enable_out_clock':
/home/lee/work/12.11/esp32-cam-demo/components/camera/./xclk.c:14:15: error: 'ledc_timer_config_t {aka struct <anonymous>}' has no member named 'duty_resolution'
     timer_conf.duty_resolution = 1;
               ^
/home/lee/esp-idf/make/component_wrapper.mk:242: recipe for target 'xclk.o' failed
make[1]: *** [xclk.o] Error 1
/home/lee/esp-idf//make/project.mk:448: recipe for target 'component-camera-build' failed

FlyingArm, 2017-12-11:
The idf version is not correct.

LilyGO, 2017-12-11:
Thank you for your reply. But most of my programs can compile and upload successfully, and this is esp-idf just built a month ago, this program has the latest version of idf it? Or I understand what you mean wrong.

LilyGO, 2017-12-11:
The problem is solved, thank you very much

closed 2017-12-22 by igrr

BasselZaity, 2018-02-14:
@LilyGO Can you please tell us how you solved the problem ?