igrr/esp32-cam-demo #57: OV2640 probe failed with error

LilyGO, 2017-12-11:
E (273) camera_demo: Camera probe failed with error 0x20003

LilyGO, 2017-12-11:
[image]

Oitzu, 2017-12-11:
0x20003 is Camera not supported.
Two causes can lead to this error:
1.) Camera-Model not activated in config.
2.) Not the correct answer over SCCB from the camera.

I guess number 1. ;)

igrr, 2017-12-11:
Might be related to https://github.com/igrr/esp32-cam-demo/issues/37. See if toggling power to the camera helps.

LilyGO, 2017-12-12:
[image]

LilyGO, 2017-12-12:
When I recompile upload, the error was upgraded. My XCLK pin is 21. I do not understand why something goes wrong. I just started getting in touch with ESP32 last month, but I want to figure it out. This looks cool. thank you all.

LilyGO, 2017-12-12:
@Oitzu hi,Oitzu.Camera model is not activated in configuration. Where to modify the configuration. Is it here, my module is ov2640

LilyGO, 2017-12-12:
[image]

igrr, 2017-12-21:
Given the log posted in https://github.com/igrr/esp32-cam-demo/issues/61, i think this issue is solved.

closed 2017-12-21 by igrr

LilyGO, 2017-12-21:
Yes, thank you for your help.Adherence is victory.

9lash, 2018-12-15:
**It looks like this issue is closed but doesnt seem to work for me. I get the following output:** 

[image]

**Pin configuration is as follows:** 
[image]

To flash the firmware, I connected GPIO0  to GND.

Oitzu, 2018-12-15:
Unrelated to this problem.
Maybe follow #106.