igrr/esp32-cam-demo #58: Add button fuction but there is something wrong

donny681, 2017-12-12:
Eh,I want to add a fuction that the camera take photo after pressing a button.I refer to esp-idf->peripherals->gpio demo and found  that gpio_install_isr_service   is incompatible with gpio_isr_register().So sad.
Do you have a plan to add a button function.

krzychb, 2017-12-12:
Maybe you can try [button component](https://github.com/espressif/esp-iot-solution/tree/master/components/general/button) from espressif/esp-iot-solution?

donny681, 2017-12-14:
@krzychb Eh,it doesn't work because the demo contains gpio_install_isr_service().The demo of espressif/esp-iot-solution also registers the isr which  is incompatible with gpio_isr_register().If wanting to use button,it's better to register the isr.I'm trying to revise  the camera demo.

closed 2017-12-21 by igrr

igrr, 2017-12-21:
This example now also uses gpio_install_isr_service, so you should be able to add button functionality to the application.