igrr/esp32-cam-demo #60: wifi send and camera data acquisition can not work at the same time?

FlyingArm, 2017-12-15:
When the camera output signal when the normal work of wifi data transmission will result in a high failure rate,if pull up the pin pwdn on the camera into the power off wifi send data can have a good effect.
Wifi send data when camera output signal will lead to wifi send problems?

igrr, 2017-12-21:
Could you please clarify "high failure rate"? This example transfers captured image over TCP, are you observing some errors in TCP transmission?

jixunx, 2019-05-20:
I have met the same issue.It's probably a hardware problem.Attempts can be made to use external crystal oscillators to provide clocks, or to use crystal oscillators of various frequencies.

RoyHsu67, 2019-08-20:
I have a similar issue.  If I only sent a image over TCP by esp32 WiFi module, the transfer was quite smooth; but when I applied the esp32 camera to get a photo and then transfer it over the same network setting. I just found the networking speed became so slow. The improvement I try was to do "initialize the camera driver", "get a camera image buffer", then immediately "deinitialize the camera driver", so after "deinitialize the camera driver", I could transfer image over TCP with normal networking speed, but it seems not the ideal solution. I ever tried to  pull up the pin pwdn on the camera, but nothing changed. I also observed the TCP connection when it was slow; there's no error message, but  a lot of false from client.available(), if I sent a bunch of data from the other end.  I also tried to use two cores of ESP32 to seperate getting camera image and WiFi TCP transmission, but nothing became faster! Any suggestion?