igrr/esp32-cam-demo #61: web server no image. But the serial port data is normal.

LilyGO, 2017-12-20:
[image]
[image]
[image]

lss6378, 2017-12-21:
@LilyGO try http://192.168.0.108/jpg_stream

LilyGO, 2017-12-21:
@lss6378 Of course I also tried.Interestingly, the serial port information in the crazy transmission of data. But web server no image

[image]

Transmission speed is fast

igrr, 2017-12-21:
Such long wires between camera and ESP are guaranteed to not work. 

Please try OV7725 in BMP (RGB) mode first. You will see the stream.

LilyGO, 2017-12-21:
Will BMP (RGB) mode how to enter？Make menuconfig.

igrr, 2017-12-21:
Connect OV7725 camera, then in app_main.c change `CAMERA_PIXEL_FORMAT`:
```
#define CAMERA_PIXEL_FORMAT CAMERA_PF_RGB565
```

LilyGO, 2017-12-25:
Before I started OV2640 module, I have finished and successfully carried out all OV7725 tasks. This is to thank everyone for your dedication. As you said, the reason my current module can not be imaged is that I guess the wiring is too long. But things go wrong. I reconnect, as before, can not display pictures and streams. Oh, right. I did not tell you before my pixel is 2 million, I speculate that there may be too large picture pixels, the picture is not finished the task is finished. If so, what should I do?

LilyGO, 2017-12-25:
[image]
[image]