igrr/esp32-cam-demo #64: Frame rate always the same on QVGA , VGA or SVGA

wdouglas1999, 2018-01-18:
I am using OV2640 at 20Mhz XCKL with great success. Jpeg streaming is fairly good at about 7 fps.

The issue i have is when i change from VGA to QVGA i would expects the frame rate to increase, but the serial monitor still outputs that the image took about 140ms to complete. 
Since QVGA is less pixels than VGA i was expecting the time taken to load the image to reduce to about half the time, say 70ms.

Can anybody help me by explaining why this is happening? 
THX

easeway, 2018-01-19:
See issue #63:

https://github.com/igrr/esp32-cam-demo/blob/master/components/camera/ov2640.c#L462:

```
    if (framesize <= FRAMESIZE_SVGA) {
        clkrc =0x83;
        regs = svga_regs;
    } else {
        clkrc =0x87;
        regs = uxga_regs;
    }
```

You will need to change `clkrc = 0x83` to `clkrc = 0x81` to raise to 15fps, and `0` or `0x80` to raise to 30fps.

wdouglas1999, 2018-01-19:
Thank you for the reply.

When I change clkrc = 0x83 to clkrc = 0x81 or 0x80. I cannot see to get any picture or stream with Xclk = 20mhz. Only a blank screen. 
I get a stream with Xclk = 5Mhz but then image take 300ms to complete. 
Any advice?

easeway, 2018-01-19:
Can you check the original data transferred from I2S, before processed by dma_filter_task? As I encountered different sample mode with different XCLK freq and different CLKRC. According to my test, when XCLK is low (like 12MHz, or lower) and lower frame rate, like 8fps, the sampling mode is 00 0A 00 00, 00 0B 00 00, while when XCLK and CLKRC is raised, the sampling mode becomes 0A 0B 0B 0C.

Please try to change sample_mode by hard code the return value of `is_hs_mode`: https://github.com/igrr/esp32-cam-demo/blob/master/components/camera/camera.c#L87
and see if things change.

wdouglas1999, 2018-01-19:
@easeway 
I only seem to be able to receive images with CLKRC = 81 when XCLK = 5MHZ. Anything above 5MHZ i receive a blank page. When i use  CLKRC = 83 i can set XCLK = 20MHZ with great pictures. 
Not sure how to  check the original data transferred from I2S, before processed by dma_filter_task.
I did however find this post which may be related.

https://github.com/igrr/esp32-cam-demo/issues/29
"
@igrr When XCLK > 10MHz, is_hs_mode() returns true, and the sampling mode is SM_0A0B_0B0C instead of SM_0A00_0B00, right? However, I receive corrupted JPEG data when XCLK > 10MHz with OV2640 camera, i believe it is due to the sampling mode. Do you have any idea to fix this? Thanks
--
"

wdouglas1999, 2018-01-20:
@easeway 
This is what is have in the serial monitor for first 16 bytes with clkrc = 0x83 and pclk = 20MHZ. Clear image on the screen.
for multipart/x-mixed-replace stream (use with JPEGs)[0m
[0;32mI (10078) camera: Data d8 00 ff 00 ff 00 d8 00 e0 00 ff 00 00 00 e0 00[0m

BasselZaity, 2018-02-14:
@wdouglas1999 I am trying to connect the same camera OV2640 , but the connection failed, do you have any idea what I can do ?
I (0) cpu_start: App cpu up.
I (319) heap_init: Initializing. RAM available for dynamic allocation:
I (326) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (332) heap_init: At 3FFB8E78 len 00027188 (156 KiB): DRAM
I (338) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (345) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (351) heap_init: At 4008EA8C len 00011574 (69 KiB): IRAM
I (357) cpu_start: Pro cpu start user code
I (152) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (235) camera_demo: Camera probe failed with error 0x20003

wdouglas1999, 2018-02-15:
@BassilZaity

I used working code from kodera2t. All connections are the same as the reference of igrr.

https://github.com/kodera2t/esp32-cam-demo

Dema323, 2018-09-15:
How do you change camera resolution ?