igrr/esp32-cam-demo #67: Ideas on "remote video monitoring" demo based at ESP32 and ov7670?

huaxiaozhong1, 2018-02-26:
Greeting!     We are planning to have a demo on “remote video monitoring” with cloud. What kind of development board would you like to recommend? ESP-wrover-kit? If yes, could you share us with more ideas? 1) Could you recommend an ov7670 camera? 2) As for esp-idf and Arduino that Espressif links to, are both all good for the implementation? 3) Looking at https://github.com/igrr/esp32-cam-demo/issues/29, esp-idf includes driver and app for ov7670, could you confirm about it? Thanks!

wdouglas1999, 2018-03-01:
Hi,
I am no expert , but from what i can see from the data sheet of the OV7670 i do not see any jpeg output. This would imply that image sizes will be bigger meaning lower frame rate. You might be better of using OV2640.

huaxiaozhong1, 2018-03-07:
Thanks for feedback! I noticed that esp-wrover-kit v3 declares the supporting for ov7670 (http://esp-idf.readthedocs.io/en/latest/get-started/get-started-wrover-kit.html). Could you let me know which format of data your board expects to receive from the 0v7670 connected?

igrr, 2018-03-07:
Common OV2640 and OV7670 modules have similar pinouts, so both are compatible with WROVER-KIT. As @wdouglas1999 has mentioned, OV7670 does not support JPEG format, so if you need to upload images to the cloud, you might be better off with the OV2640 which does.