igrr/esp32-cam-demo #69: Does the DMA memory need to be specially malloc'd?

hopkinskong, 2018-03-01:
Current implementation seems only using `malloc`, but doesn't the DMA memories need to be allocated by `pvPortMallocCaps(size, MALLOC_CAP_DMA)`?

I am running BLE+WI-Fi+A I2C task, and I got heap corruption when i visit the JPEG stream webpage, I am wondering if is it the issue of incorrect malloc.

hopkinskong, 2018-03-03:
The memory issue I suppose is caused by low stack size in `xTaskCreate` (yet to be confirmed).

But I am still interested in the DMA malloc question. Does DMA memories need to be allocated differently? Of course, the framebuffer memory can be allocated directly with `malloc`, and thus, could reside in PSRAM, however, shouldn't the DMA line buffers be allocated differently as it cannot be reside in PSRAM?

@igrr Any insights on my question?

AndrewGi, 2018-04-17:
According to http://esp-idf.readthedocs.io/en/latest/api-guides/external-ram.html, DMA can't write to PSRAM. I do believe the framebuffer memory could be allocated in PSRAM