igrr/esp32-cam-demo #79: Can it work if I fix VSYNC, HSYNC and H_ENABLE signals to 1 and provide BCK clock?

HuyK, 2018-06-05:
Hi all,

I want to control I2S receiving in Camera mode such that all 3 signals VSYNC, HSYNC and H_ENABLE are fixed to 1 by GPIO Matrix commands and make I2S receives data whenever BCK clock applied.
I tried to modify the camera demo in  gpio_init() function to fixed HIGH to these three signals:
    gpio_matrix_in(0x38, I2S0I_H_SYNC_IDX, false);              //fix HSYNC to 1
    gpio_matrix_in(0x38, I2S0I_V_SYNC_IDX, false);              //fix VSYNC to 1
    gpio_matrix_in(0x38, I2S0I_H_ENABLE_IDX, false);          //fix H_ENABLE to 1
   gpio_matrix_in(CAM2_PCK, I2S0I_WS_IN_IDX, false);

By doing this, I expect to only give PCK clock and D0-D7 data to I2S and it can receive data normally. But actually, I see there is no I2S interrupt event when I provide PCK clock.
Is it possible to have I2S working that way? Meaning fix HIGH to both three control signal and only provide BCK clock and Data to I2S?

Thanks,
Huy

HuyK, 2018-06-05:
The purpose of changing that way is that I want to adapt I2S receiving part to OV7670 FIFO module. Currently, some demo code using GPIO bit-bang mechanism to provide RCK clock to read FIFO data but the max frequency is about 1.7MHz which is quite slow. I plan to use an external oscillator to provide about 10MHz to both FIFO's RCK and I2S's PCK to read out data to ESP32's RAM through I2S and DMA.

nkolban, 2018-06-24:
I think what you suggest should work.