igrr/esp32-cam-demo #80: Latest esp-idf master breaks working camera code that uses the ledc submodule

igrr, 2018-06-06:
_From @buckybots on June 6, 2018 1:44_

By reverting esp-idf back to earlier commit, the error does not happen.

The earlier commit where the code was working is :
$ git show
commit 7d13e3fa6af8c500ac207b2734c2611cf2f24e34 (HEAD)
Merge: 22ae5adb 37b98f73
Author: Angus Gratton <angus@espressif.com>
Date:   Wed Feb 14 09:45:09 2018 +0800

Bringing esp-idf up to date with:
Git checkout master
Git pull
Git submodule update --init
$ git status
On branch master
Your branch is up to date with 'origin/master'.
nothing to commit, working tree clean
$ git show
commit b2ff235bd003bf3dcb114ff7efe3e0c605c4dd9f (HEAD -> master, origin/master, origin/HEAD)
Merge: 8a4c9d7d a8504b96
Author: Angus Gratton <angus@espressif.com>
Date:   Tue Jun 5 11:51:35 2018 +0800

Now flashing code with latest esp-idf causes the following errors:  (The code uses igrr’s camera code, and appears to be breaking when using the ledc code to create the XCLK PWM output.

Here is info from monitoring the output :

Flashing binaries to serial port COM3 (app at offset 0x10000)...
esptool.py v2.3.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
…….
D (1202) intr_alloc: Connected src 22 to int 13 (cpu 0)
D (1202) camera: Enabling XCLK output
**E (1202) ledc: ledc_set_duty_with_hpoint(383): hpoint argument is invalid**
D (1207) ledc: LEDC_PWM CHANNEL 0|GPIO 19|Duty 0001|Time 0
D (1213) camera: Initializing SSCB
D (1217) camera: Resetting camera
D (1240) camera: Searching for camera address
**E (1376) test.c: Camera probe failed with error 0x20001**

_Copied from original issue: espressif/esp-idf#2032_

igrr, 2018-06-06:
Moved here because this is a bug in this project, not IDF. `camera_enable_out_clock` function should zero-initialize `ledc_timer_config_t` and `ledc_channel_config_t` structures.

vivian-ng, 2018-07-02:
Any suggestions on how this can be solved?

vivian-ng, 2018-07-03:
I added
`ch_conf.hpoint = 0;`
just before
`err = ledc_channel_config(&ch_conf);`
in `esp_err_t camera_enable_out_clock(camera_config_t* config)` found in xclk.c and was able to get the program to run.

ModMike, 2018-09-06:
I have the same issue and it looks like the code is already there.  Any other suggestions?

ModMike, 2018-09-06:
I am able to get it to compile if I comment out `ch_conf.hpoint = 0;`  Will this have any negative consequences?

janoist1, 2018-09-19:
What is `hpoint`?

Gabones, 2019-05-25:
E (6979) ledc: ledc_set_duty_with_hpoint(383): hpoint argument is invalid

anyone can help me to solve this problem?

I tried to set ch_conf.hpoint = 0;
before
err = ledc_channel_config(&ch_conf);
  
but it didn't work