igrr/esp32-cam-demo #81: Simple function to reduce motion smearing by reducing OV2640 shutter speed

s60sc, 2018-06-12:
Because the OV2640 on the ESP32 runs at limited frame rates, the auto exposure causes motion smearing. Reducing the shutter speed in good light reduces smearing. So I added the simple function below to my copy of ov2640.c

`static int set_shutter_divider (sensor_t *sensor, int shutterDivider) {
  /* from http://www.arducam.com/manual-exposure-ov2640
     the minimum exposure time is 1 Line Time, and the maximum exposure time is one Frame time.
     mimimum exposure time depends on pclk (53.4uSecs @ 36MHz)
     pclk = xclk /(decimal value of clkrc[0:5]+1. Bit 7 is frequency doubler
     Required register value is a multiplier of the minimum exposure time
     Exposure time multiplier is spread across 3 registers.
     However:
     This function just reads the current value of the middle register and divides it down to reduce shutter time for faster exposure, or increase it for low light
     Must disable automatic exposure control beforehand:
     s_state->sensor.set_exposure_ctrl(&s_state->sensor, 0); // disable AEC
     s_state->sensor.set_shutter_divider(&s_state->sensor, 2); // reduce shutter speed to a quarter of auto
  */
  int ret=0;
  uint8_t AECreg = SCCB_Read(sensor->slv_addr, AEC); // provides bits 2-9 of exposure time multiplier
  uint8_t newAECreg = AECreg>>shutterDivider; // shutterDivider is a +ve or -ve bit shift value
  if (newAECreg < AECreg && shutterDivider < 0) return ret; // abandon if increased shutter time would overflow register
  else ret |= SCCB_Write(sensor->slv_addr, AEC, newAECreg);
  return ret;
}`

wdouglas1999, 2018-07-02:
@s60sc
Hi, I too have noticed this smearing and would like to know.
Is this script working? How do I implement it?
Or is this a question we should work on?
Thx

wdouglas1999, 2018-09-28:
I managed to get this to work for a single image, but how would I use this to control the exposure while streaming video? When I turn the auto exposure off I cannot seem to turn it in again.
Must the camera module be reset every time for this? 
Any help will be appreciated.
Thx