igrr/esp32-cam-demo #84: Change I2S pins

enricolam, 2018-06-21:
Can anyone please tell me where in the code can I change the default I2S pins to other gpios?

james-lan, 2018-06-24:
When you do the make menuconfig you can change the pins in:  ESP32 Camera Demo Configuration  --->   Pin Configuration  --->

ats3788, 2018-06-26:
Hello
I thought that I2S Pins are fixed connected to DAC_1 GPIO25 and DAC_2 GPIO26
 Vertical Sync | VSYNC | IO25 | IO22 |
 Horizontal Reference | HREF | IO23 | IO26 |
Why I can alternate this pins ?

enricolam, 2018-06-26:
In my knowledge, those are default pins, but shouldn't be fixed. Esp32 have a gpio matrix according to what I know. Which makes it possible for us to reroute the i2s to other pins. But I might be wrong, and please point it out for me if I have mistaken anything.