igrr/esp32-cam-demo #98: Is it possible to take full resolution images from ov2640?

netremo, 2018-08-20:
OV2640 is 2MP module, but currently in this example maximum frame size is 800x600. Is it possible to take full resolution photo? I need only one shot image.

Thanks.

lss6378, 2018-08-21:
I think it's possible if you use an esp32-wrover module. It has a 4M PSRAM module.

tmrttmrt, 2018-09-02:
Take a look at https://github.com/tmrttmrt/esp32-cam-demo/tree/tmrt. With enough compression it is possible to obtain jpg images with sizes of about 100 kB.

tmrttmrt, 2018-09-07:
I am new to esp32 so have no idea.

sysizlayan, 2018-11-05:
> @tmrttmrt I downloaded your repo and tried to compile but get this error:
> 
> xclk.c:31:12: error: 'ledc_channel_config_t {aka struct }' has no member named 'hpoint'
> ch_conf.hpoint = 0;
> I understand this happened as the result of the new version of ESP-IDF. I commented out the line but not sure if it will break something else

I am building with IDF3.2 latest release. I think if you update the version, it will compile.

And I am also wondering if we can take the snapshots with 1600*1200 resolution.
Memory allocation of the frame buffer is done using calloc function which allocates memory from internal SRAM, does not use PSRAM. In addition to that, even if the buffer is allocated in PSRAM using MALLOC_CAP_SPIRAM flag, this memory buffer could not be read from or written into by DMA.

I2S uses DMA to write the data to buffer, so PSRAM seems impossible to use for frame buffer to me.

Ping-pong buffering may be a solution but I am not sure if we can change the buffer address while the I2S is reading data from the camera.

@igrr do you have any comments on this issue? If you can inform us about the issue, it would be great.

Thanks in advance.

javiercuellar73, 2020-03-23:
> Short answer: Yes it is possible. No problem at all with the 4M PSRAM module.

Do you have the long answer? How to shoot at 1600*1200 resolution??