igrr/esp32-cam-demo #99: Pictures are scrambled

ModMike, 2018-09-06:
I am using the M5stack esp32cam with your code because their fork has a compilation error and I had to comment out a line.

My use is case is fairly light.  I am using the camera to monitor a 3D printer.  I don't need high res or super fast frame rate, although 15 fps would be best.

I am running into 2 issues:

1) When the module is first plugged in, I get video but it has a bit of a strobe effect.  Is there any way to correct this?

2) Fairly rapidly, within a minute or 2, the camera starts sending de-synched images.  Not sure if that is the right term but it looks like the picture was cut into vertical strips and is not being assembled properly.

I have 3 of these modules.  The first 1 burned up and the other 2 have the same issues.  Any advice?

I also found a lot of forks but am not sure who has most active one.  Which one do you recommend?

Thank you.

tmrttmrt, 2018-09-10:
I have played with such module. I was able to get stable pictures with decreasing xclk below 10 MHz. I believe that the ESP32 clock stability is an issue for these modules.   
For the best throughput at lower xclk one needs also to modify some camera registers that control the internal camera pclk. Take a look at tmrt branch at tmrttmrt/esp32-cam-demo that works with 8 MHz xclk.

ilovehotmilk, 2018-09-13:
@tmrttmrt 
I found this in your code. `#define CAMERA_FRAME_SIZE CAMERA_FS_UXGA`
What is FPS at this resolution?
And can I improve FPS if I lower the resolution?