igrr/esp32-cam-demo #104: error: unknown type name 'http_context_t'

mau-lopez, 2018-10-30:
Hi, just cloned esp32-cam-demo, and while making I get the attached compiler errors, all related with http_context_t, has anybody had the same or similar error? or any idea of how to solve it? Thanks in advance

Macbook Pro Mid 2015, MacOS 10.14
esp-idf cloned on Oct 26, 2018

[image]

fah, 2018-11-03:
Same here. Replaced the include of "http_server.h" by "esp_http_server.h" then got the messages above. Found no solution.

fah, 2018-11-04:
Solution: (tested on MacOSX)
1. go into components folder
1. remove http-server folder
1. execute ````git clone https://github.com/igrr/esp32-http-server.git````
1. go back to root of project and execute ````make flash monitor`````

mau-lopez, 2018-11-05:
Thanks a lot for your help fah, indeed it worked, no more compiling errors. Closing this issue now.

closed 2018-11-05 by mau-lopez

WJohnnyW, 2019-02-17:
it really works! Thank you from the bottom of my heart!

treii28, 2019-07-20:
thanks, ran into this today