igrr/esp32-cam-demo #106: camera probe failed (0x20001) with OV7670

Skel0t, 2018-11-28:
Hello, so I tried to get the OV7670 running on the Heltec Wifi Kit 32. 

The problem is I'm getting following Code spit out when flash monitoring:

`I (29) boot: ESP-IDF v3.3-dev-236-gfa59b1b1c 2nd stage bootloader
I (29) boot: compile time 13:12:39
I (37) boot: Enabling RNG early entropy source...
I (38) boot: SPI Speed      : 80MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18a80 (100992) map
I (122) esp_image: segment 1: paddr=0x00028aa8 vaddr=0x3ff80000 size=0x00000 (     0) load
I (123) esp_image: segment 2: paddr=0x00028ab0 vaddr=0x3ff80000 size=0x00000 (     0) load
I (129) esp_image: segment 3: paddr=0x00028ab8 vaddr=0x3ffb0000 size=0x030c0 ( 12480) load
I (142) esp_image: segment 4: paddr=0x0002bb80 vaddr=0x3ffb30c0 size=0x00000 (     0) load
I (147) esp_image: segment 5: paddr=0x0002bb88 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/Woshi/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (156) esp_image: segment 6: paddr=0x0002bf90 vaddr=0x40080400 size=0x04080 ( 16512) load
I (171) esp_image: segment 7: paddr=0x00030018 vaddr=0x400d0018 size=0x6b394 (439188) map
0x400d0018: _stext at ??:?

I (301) esp_image: segment 8: paddr=0x0009b3b4 vaddr=0x40084480 size=0x0d67c ( 54908) load
0x40084480: dma_filter_grayscale_highspeed at C:/msys32/home/Woshi/esp/esp32-cam-demo/components/camera/camera.c:742

I (321) esp_image: segment 9: paddr=0x000a8a38 vaddr=0x400c0000 size=0x00000 (     0) load
I (321) esp_image: segment 10: paddr=0x000a8a40 vaddr=0x50000000 size=0x00000 (     0) load
I (328) esp_image: segment 11: paddr=0x000a8a48 vaddr=0x50000000 size=0x00000 (     0) load
I (347) boot: Loaded app from partition at offset 0x10000
I (348) boot: Disabling RNG early entropy source...
I (349) cpu_start: Pro cpu up.
I (352) cpu_start: Starting app cpu, entry point is 0x40081068
0x40081068: call_start_cpu1 at C:/msys32/home/Woshi/esp/esp-idf/components/esp32/cpu_start.c:246

I (0) cpu_start: App cpu up.
I (363) heap_init: Initializing. RAM available for dynamic allocation:
I (370) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (376) heap_init: At 3FFB9070 len 00026F90 (155 KiB): DRAM
I (382) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (388) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (395) heap_init: At 40091AFC len 0000E504 (57 KiB): IRAM
I (401) cpu_start: Pro cpu start user code
I (194) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
**E (269) camera_demo: Camera probe failed with error 0x20001**
`

I researched many topics here on github and also found https://github.com/igrr/esp32-cam-demo/issues/29 
I've read something about that this error means "not connected", but I checked my wiring multiple times, even redid it a few times and also changed pins in case that would be the problem. The camera should be connected but still I'm not getting rid of this error.
Also, in that thread everyone seems to be able to get the camera running, although there are some bugs, but for me it does not even go that far...

Is there anything I can do to get it working properly?

sige1, 2018-12-11:
It's the same Error 0x20001 with OV2640.
It seems there is a problem with software.

TheNitek, 2018-12-11:
It just means not connected. Most probably an issue with your wiring or the pin configuration.

Skel0t, 2018-12-11:
I'm using the "Alternate ESP32 Pin Mapping" specified in the readme and changed it according to it in the menuconfig.

I'm using the esp32 DevKitC (find pinout here: https://cdn.shopify.com/s/files/1/1509/1638/products/Pinout_Darstellung_ESP32_Developmentboard_339d0add-2031-4bca-88af-026cbbd0656d.png?v=1520380031 ) and checked my wiring multiple times. I'm still receiving this error.
I already thought the camera might be defect?

Oitzu, 2018-12-15:
> It just means not connected. Most probably an issue with your wiring or the pin configuration.

More precisely: The software tries to determine the camera model by asking via SCCB for the camera id.
If no SCCB connection can be initialized the error 0x20001 (ESP_ERR_CAMERA_NOT_DETECTED) occurs.

As others mentioned: Check the connections.
Sometimes there can be a problem with the reset cycle on the camera, try following:
Unplug the setup. Hold down the reset button on the esp and reconnect.
While still holding the reset button open the Serial-Terminal connection.
Release the reset button and watch for output.

Skel0t, 2018-12-15:
Thanks for your help. For me the problem was fixed by using another camera, seems like my old one was defective. I ordered the same camera twice again and tried it out and it instantly showed me 0x20003, and so with adaption to the code it worked for me then.

sige1, 2018-12-20:
0x20001 error remains with the combination of m5stack m5camera and ov2640.
Which pin assignment should I compile?

below is 0x20001.
CONFIG_D0=17
CONFIG_D1=35
CONFIG_D2=34
CONFIG_D3=5
CONFIG_D4=39
CONFIG_D5=18
CONFIG_D6=36
CONFIG_D7=19
CONFIG_XCLK=27
CONFIG_PCLK=21
CONFIG_VSYNC=22
CONFIG_HREF=26
CONFIG_SDA=25
CONFIG_SCL=23
CONFIG_RESET=15

BR.

caetanowagner, 2019-02-21:
My first problem was pinout connection. After that I changed few lines in code.
What I did:

added files
ov7670_regs.h
ov7670.c
ov7670.h
I copied files from ov7725 and modified it

add in camera.c lines
#if CONFIG_OV7670_SUPPORT
#include "ov7670.h"

and lines
    switch (id->PID) {
#if CONFIG_OV7670_SUPPORT
        case OV7670_PID:
            *out_camera_model = CAMERA_OV7670;
            ov7670_init(&s_state->sensor);
            break;

 in file ov7670.c add lines
#include "ov7670.h"
#include "ov7670_regs.h"

modified line int ov7725_init(sensor_t *sensor) to ov7670_init(sensor_t *sensor)
 

in app_main.c add new if statement after CAMERA_OV2640 like the code below
    if (camera_model == CAMERA_OV7725)
    {
        s_pixel_format = CAMERA_PIXEL_FORMAT;
        camera_config.frame_size = CAMERA_FRAME_SIZE;
        ESP_LOGI(TAG, "Detected OV7725 camera, using %s bitmap format",
                CAMERA_PIXEL_FORMAT == CAMERA_PF_GRAYSCALE ?
                        "grayscale" : "RGB565");
    }
    else if (camera_model == CAMERA_OV2640)
    {
        ESP_LOGI(TAG, "Detected OV2640 camera, using JPEG format");
        s_pixel_format = CAMERA_PF_JPEG;
        camera_config.frame_size = CAMERA_FRAME_SIZE;
        camera_config.jpeg_quality = 15;
    }
    else if (camera_model == CAMERA_OV7670)
     {
         s_pixel_format = CAMERA_PIXEL_FORMAT;
         camera_config.frame_size = CAMERA_FRAME_SIZE;
         ESP_LOGI(TAG, "Detected OV7670 camera, using %s QVGA format",
                 CAMERA_PIXEL_FORMAT == CAMERA_PF_GRAYSCALE ?
                         "grayscale" : "RGB565");
     }
     else
    {
        ESP_LOGE(TAG, "Camera not supported");
        return;
    }

Maybe you can have another issues after that but will ease to follow errors and warnings from eclipse and get compile done.