igrr/esp32-cam-demo #109: OV2640 jpeg image with pixelated edges

JPMJPM, 2019-01-05:
Hi,

I have developed a OV2640 camera JPEG streaming solution with :

* Over Ethernet.
* Only two I2S 2.800 32 bits words buffers (ring buffer).
* There is not frame buffer.
* 50 FPS.
* SVGA.

Everything seem to be OK but the image edges are pixelated.
Can anyone tell me what may be happening ?.
I attach TXT (C) and JPEG files.

Thanks in advanced and regards. 

[ov2640.txt](https://github.com/igrr/esp32-cam-demo/files/2729738/ov2640.txt)

[image]

amenimtibaa, 2019-03-20:
Hello, I am trying to make your code work with my esp32 wrover and my ov2640.
Can you please tell me which board did you use ? and the connection with your camera ?

JPMJPM, 2019-03-21:
I used a ESP32-WROOM module on a custom board.

The connection between ESP32 module and camera is in the source code file.