igrr/esp32-cam-demo #114: 0v7725 rgb565

pmpz, 2019-03-04:
other than changing app_main.c

#define CAMERA_PIXEL_FORMAT CAMERA_PF_GRAYSCALE
to
#define CAMERA_PIXEL_FORMAT CAMERA_PF_RGB565

what do I need to do to get rgb image in webserver?
and I get this error after changing that line

I (148) camera_demo: Detected OV7725 camera, using RGB565 bitmap format
E (238) camera: Failed to allocate frame buffer
E (238) camera_demo: Camera init failed with error 0x101

xautzxc, 2019-03-06:
the same to you ,how do you solve it？

yeapkrush, 2019-04-29:
Hi, I'm also not an expert on ESP32 or CAM interfacing.
But it seems currently the library doesn't support RGB565 in OV7725. 
To fix the issue you're having, all you need to do is enable the SPI ram in menu config. 
After all this still I get three images overlay on each other, maybe because it still doesn't support RGB565.