igrr/esp32-cam-demo #117: Red Board OV2640

clcorreg, 2019-03-23:
Have someone any suggestion in order to use the red board ov2640? I mean the one without xclk, but with a 12MHz clock inside, and a NC pin. The pics are visible, but obviously out of sync.

clcorreg, 2019-03-31:
The image was forming but it was out of sync. Since XCLK is needed only to "startup" the camera shooting process, I supposed that my problems were in 2 places: the wiring or, possibly, that the program could use its internal xclock genertation to expect a pclock or other sync behaviour. 
So, I applied 3 distinct "patch":
1) 3 pull-up resistor to hsync, vsync, pclock;
2) shut down the internal xclock: I modified camera.c, line 130, to camera_disable_clock();
3) in menuconfig, put xclock freq to 0.
I suspect that the more important is the last one.

Now pics are stable and I can also stream SVGA to 3/5 fps.

closed 2019-04-03 by clcorreg

maciekczwa, 2019-10-23:
@clcorreg 
I have problems getting images from this module, could you tell me the exact wiring that you have used?

clcorreg, 2019-10-23:
It depends on the esp module: I used the alternate pin mapping. Anyway, you
can somehow change it via configurator.

maciekczwa, 2019-10-24:
I have generic wrover board.

Did you connected pwdn, reset?

I'm trying with following pin mapping:
io4 - d0
io5 - d1
io18 - d2
io19 - d3
io36 - d4
io39 - d5
io34 - d6
io35 - d7
io25 - vsync
io26 - sda
io27 - scl
io15 - reset
io22 - dclk
io23 - href

clcorreg, 2019-10-24:
I did it on a wroom. I also have a wrover/psram, but when I upload the code
it resulted in a failure. Didn't had the time to fix it, 'til now.

Anyway: I didn't connect only pwdn.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>